knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #75: [BUG] Wrong Mesh leveling grid applied in Print! (megaX)

J0nny12, 2021-02-04:
### Bug Description

Hi, unfortunately after 3 days of trying to figure it our by myself i need your help.
I can probe the bed with the BL touch using G29, the mesh is taken correctly, Mesh visualizer in Octoprint shows that.
HOWEVER in the actual print it's ignored by the printer!
Doesnt matter if special menu leveling or G29 via octoprint.
It seems to use an alternative, WRONG or OLD Mesh. Z spindles are moving

any help will be appreciated!

### My Configurations

MEGA_X_TMC_BLT_10_v1.1.9.hex
i got
G28
G29
M420 S1

as start Gcode

### Steps to Reproduce

1. run G29
2. Print and watch the layers go from way too thin to way too thick

**Expected behavior:** 

Correct Mesh being applied, nice first layer

**Actual behavior:** 

Bad uneven first layer

OkiGraphics, 2021-02-05:
This seems like similar issue (#72 ) as I had. A workaround for me was to create a mesh and save it using Pronterface.

J0nny12, 2021-02-05:
A Mesh is created, it's just applied in a weird way.
the spindles are moving.

closed 2021-03-13 by github-actions[bot]