knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #76: [BUG] Mega X - Extruder loses steps

Sydney1101, 2021-02-07:
### Bug Description
During the strong movements when retracting the filament, the extruder motor " clicks ", which indicates the loss of steps.

### My Configurations
MEGA_X_TMC_v1.1.9.hex

### Steps to Reproduce
I have increased the voltage on the TMC2208 - but without success. From there, it may have something to do with the timing on the motor control. I also mean that the y-axis motor sounds different than usual - but it does not lose steps.

**Expected behavior:** [What you expect to happen]
no steps will be lost

**Actual behavior:** [What actually happens]
some steps will be lost

#### Additional Information
n.a.

Sydney1101, 2021-02-09:
Maybe it is necessary to switch to UART mode to enable spread cycle mode on the extruder driver or to use the old driver (only for the extruder)...

Moskito99, 2021-02-12:
Did you calibrate the extruder?
My Mega X has the wrong steps from the beginning and thats why my extruder was clicking.
The default extruder steps are 400 which is too much.
I changed it to 390.53 after calculation... 
https://3daddict.com/3d-printer-extruder-calibration-steps/

But, I have also the y-axis motor sound which was not there with the original Marlin software...

Sydney1101, 2021-02-13:
Hi Moskito99. Loosing steps are not a problem of calibration ;-) Currently I did some changes on my Mega X... I now use a heat sink for the extruder stepper motor and the temperature has come down. I also replaced the fans that cool the TMCs with more powerful versions, although they are now louder. I wanted to rule out the possibility that temperature is a problem. I also replaced the hotend fan with a more powerful fan, since the Noctua is quieter but doesn't cool enough. The printouts are now better. Currently, I'm not losing steps on the extruder, but the y-axis motor continues to make strange jerking noises - though I haven't made any complex prints yet. So I will observe the problem at the extruder. With the noises of the y-axis (especially with 45°-hatchings and circles) I still have not made any progress.

Translated with www.DeepL.com/Translator (free version)

Moskito99, 2021-02-13:
Hi Sydney1101,
you need a heat sink for your stepper motor? Your voltage is to high!
I have done the same tuning as you and a lot more...
If the extruder try to extrude more than the nozzle can extrude, the extruder loosing steps and you hear the extruder clicking.
What is your extruder stepping? The default 400 is to much, so you must change it!

Sydney1101, 2021-02-13:
Meine Spannung beträgt aktuell 1.1 V (TMC 2208). Meine Schritte passen zu meinem aktuellen Dual-Antrieb-Extruder (412). Mit dem Original-Extruder hatte ich auch so um die 392 Schritte. Abhängig vom Filament verändere ich den Extrusionsfaktor, um Überextrusion zu vermeiden. Die Extruder-Motortemperatur geht nur hoch, wenn ich viele Rückzüge habe. Von daher passt es nun mit dem Kühlkörper und die Temperatur beträgt so um die 40°C.

Sydney1101, 2021-02-13:
Ok. I have now compiled my own versions and tried everything possible. My conclusion: The problem comes from the Linear Advanced function - maybe in combination with the retract. After I disabled this there is no more stuttering and no other strange noises from another axis. Normally, the function should slow down smoothly rather than jerkily.

Sydney1101, 2021-02-13:
Ok. I have now compiled my own version and tried everything possible. My conclusion: The problem comes from the Linear Advanced function - maybe in combination with the retract. After I disabled this there is no more stuttering and no other strange noises from another axis. Normally, the function should slow down smoothly rather than jerkily.

[https://github.com/MarlinFirmware/Marlin/issues/16155](url)

Sydney1101, 2021-03-02:
Update. Meanwhile I changed my Mega X with TMC2208 for UART operation and compiled myself the latest version of Marlin. Now everything runs very harmoniously and without problems. Also I found out that the extruder motor should run in a mode where it is not possible to switch in standalone mode - so it has not enough power. If you can't use the UART mode, you should rather use the old motor driver for the extruder (applies to TMC2208). Other versions of TMC can be set to the correct mode (spreadcycle) by soldering bridge ;-). For the motor drivers Marlin now takes over the switching, depending on the speed, and it all runs much better.

closed 2021-04-08 by github-actions[bot]