knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #77: Anycubic Mega X mit anderem Extruder - falsche Drehrichtung

SpRiNgToIfEl, 2021-02-07:
Hallo zusammen, ich nutze deine Firmware schon eine Weile. Nun hat sich leider zwischenzeitlich der org. Extruder vom Mega X verabschiedet sodass ein neuer eingebaut wurde.

https://www.amazon.de/gp/aw/d/B07SY745CF?psc=1&ref=ppx_pop_mob_b_asin_image

Jetzt ist das Problem, das der Extruder in die falsche Richtung dreht, dh statt das Filmament zu fördern, wird des zurückgezogen.

Im Mega X sind 5x TMC2208 und der oben genannte neue Extruder verbaut.

Installiert ist v1.1.9  _X _TMC wo nun jedoch die Drehrichtung aufgrund fehlender Übersetzung am Extruder falsch ist. Wie kann man es ändern, ohne die Kabel zu tauschen? Habe auch die Marlin.ino schon bemüht, jedoch gibt es mit Arduino IDE Probleme beim Compelieren, dh es fehlen irgendwelche Dateien oder die Verweise führen ins nichts.

Auf Wunsch kann ich auch die Fehlermeldungen Posten.

Gruß Stefan

SpRiNgToIfEl, 2021-02-08:
...habe es nochmals mit VSC / PlatformIO probiert und bekomme bei Compelieren folgende Fehler:
[image]

kscheff, 2021-02-10:
Der einzige Fehler hier ist dass Compilieren mit i geschrieben wird ... ;-)
Das sind Warnungen, unschön aber eine Firmware sollte trotzdem dabei rauskommen.

tbirn, 2021-02-13:
@SpRiNgToIfEl Ich hatte das gleiche Problem (Mega X und Extrudertausch).

In der `Configuration.h` gibts den if block `#if ENABLED(KNUTWURST_TMC)` und darunter dann nochmal `#if ANY(KNUTWURST_MEGA, KNUTWURST_MEGA_S, KNUTWURST_MEGA_X)`

Dort musst du folgenden Block so anpassen:
`#if ENABLED(KNUTWURST_BMG)
            #define INVERT_E0_DIR false // set to false for stock drivers or TMC2208 with reversed connectors
        #else
            #define INVERT_E0_DIR **false** // set to false for stock drivers or TMC2208 with reversed connectors
        #endif`

closed 2021-03-20 by github-actions[bot]