knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #78: Anycubic Mega X mit neuem Displey DGUS

LosPimpos, 2021-02-08:
Hallo,

kannst du bitte eine Firmware kompilieren wo der Anycubic Mega X und das neue Display integriert sind?...
Also quasi eine MEGA_X_DGUS_v1.1.9.hex.

Steppertreiber etc ist alles noch original, nur leider ist meine Platte etwas verzogen, hoffe durch das Mesh Bed Leveling bekomme ich das irgendwie in den Griff.

Danke....

knutwurst, 2021-02-08:
Hallo,
nein, sorry, denn beim Mega X gibt es kein anderes Display als das, was bereits darin läuft. Nur und ausschließlich der Mega S verwendet zwei unterschiedliche Displays und nur hier macht die Unterscheidung Sinn. 

Installiere dir einfach die normale Mega X Firmware. Die läuft ;)

LosPimpos, 2021-02-08:
Alles Klar....

Danke :)