knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #79: [BUG] (short description)

mrvice, 2021-02-10:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step] start mesh leveling and complete it
2. [Second Step] 1 mesh leveling point to high not able to change 1 point
3. [and so on...] you have to do them all over again

**Expected behavior:** [i would expect to be able to adjust the 25 points individual without loosing the settings on all of them]

**Actual behavior:** [all mesh level points get reset if mesh leveling is started a 2nd time.]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

OkiGraphics, 2021-02-10:
I have exactly the same problem. I was not sure if its feature or bug so I did not create issue, but I would really appreciate if new mesh bed leveling took already existing mesh as a base.

kotarokun, 2021-03-17:
same here... just want to fine my Mesh-Leveling and have to leveling all point from Zero, not from last Mesh-Leveling-Point...

knutwurst, 2021-04-10:
That's not a "bug". It's the way, Marlin makes use of the "next mesh point" input. You can use an external configuration tool or a terminal to print out the mesh and edit the mesh points if you want.

Take a look at https://marlinfw.org/docs/gcode/G029-mbl.html

closed 2021-04-10 by knutwurst