knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #80: [BUG] slowing too much in smaller curves: Arc support and Ark Welder

brunoosti, 2021-02-10:
Print e Benchy using the Arc Wlder plugin (https://community.ultimaker.com/topic/34888-arc-welder-anti-stutter-plugin) to se the problem in action on the first layer with the lettering: when moving through the smaller curves the printer almost stops and almost wont extrude any filament, leaving melted and under extruded text.
Thanks for the awesome firmware : )

brunoosti, 2021-03-13:
Still happening.

hawkeyexp, 2021-03-15:
Any ideas would be nice to solve this :-)

brunoosti, 2021-03-15:
I've tested with and without linear advance and changing the Arc parameters in cura. 
Even tested with a junction deviation enabled firmware to see if Arc Welder would aid JD in some way. But had no luck.

henniero, 2021-03-17:
I had the same issue. I found this thread in the reprap forum https://reprap.org/forum/read.php?415,869738,869785
I changed line 219 in Marlin>src>gcode>motion>G2_G3.cpp

`if (!planner.buffer_line(raw, scaled_fr_mm_s, active_extruder, 0 // "0" was "seg_length" in original version`

I replaced "seg_length" with "0" at the end.
I built a new firmware with my printer-configuration from the platform.io envelope choice.
Issue fixed for me with that version.

brunoosti, 2021-03-17:
I don't know what you did, but you did it! : D

closed 2021-03-17 by brunoosti

knutwurst, 2021-03-17:
Fixed in https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/c14f93323c2bc796f90b00d7b75493d6ce05c8a6

Thank you very much @henniero !

brunoosti, 2021-03-17:
I think this "fixed" Junction Deviation. 
Making some tests right now, but the main shape that always gave me zits is completely smooth. No rattling during curved moventes and great corners. 
The only thing a bit different is that the extruder is more aggressive on the corners with linear advance.