knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #85: MEGA_X_TMC Variant crashes into endstop

boredomwontgetus, 2021-02-18:
### Bug Description

Using the MEGA_X Variant with the default stepper drivers worked perfect. 

After changing the stepper drivers to TMC2208 1.2 i flashed the MEGA_X_TMC Variant. 
With this variant the steppers navigate over the endstop on all 3 axes. Hence, "crashing" into the end with a noise indicating that the steppers kept going to far.

### My Configurations

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.1.9/MEGA_X_TMC_v1.1.9.hex

### Steps to Reproduce

1. Get an Anycubic Mega X Printer
2. Change stepper drivers to TMC2208 1.2
3. Flash Knutwurst Marlin MEGA_X_TMC Variant (https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.1.9/MEGA_X_TMC_v1.1.9.hex)
4. Power Cycle
5. Use printer display to navigate any axis to endstop

**Expected behavior:** 
All axes stop at the endstop switch.

**Actual behavior:** 
All axes go over the endstop. Pushing the directional button on the display again makes the printer still wanting to go on.

knutwurst, 2021-02-18:
That's not a bug. It's the way Marlin 2 handles the soft-endstops. You **must** do a G28 homing, so the bed and endstops are known. After that, you can use the directional keys without crashing.

closed 2021-02-18 by knutwurst

boredomwontgetus, 2021-02-18:
Mate, i am so sorry for being an idiot! :dagger: 

I did not see that anywhere your (BTW, awesome) docs.
Thank you so much.

Be sure to have a beer from the incoming donation.

knutwurst, 2021-02-18:
You are not an idiot :)
The question has actually been asked several times and I actually wanted to change the behavior. But the alternative would be that the axes cannot be moved without G28. I think that's kind of stupid too.

I will maybe update the FAQ ;)

Thank you!