knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #88: [BUG] Building error for MEGA_S_DGUS_TMC_BLT_10

m0rzan, 2021-02-21:
### Bug Description

First of all: Thanks for your great work Knutwurst!

It seems there is an issue with the building process for the MEGA_S_DGUS_TMC_BLT_10.
If I try to build it with platform io I get an error message as following:
`avr-g++: error: CreateProcess: No such file or directory
*** [.pio\build\MEGA_S_DGUS_TMC_BLT_10\firmware.elf] Error 1`

First I thought I missed an essential point in building a firmware but it seems to be a problem with the TMC Version only. Finally after two days of confusion and questioning my sanity I tried the build process with the non TMC Version where I get no errors.

### Steps to Reproduce

1. unzip the Marlin Master Folder
2. select the MEGA_S_DGUS_TMC_BLT_10 Version in platform io
2. select build
3. get error as shown above

hawkeyexp, 2021-02-21:
I have build same version 2 days ago and had the same error in WINDOWS - switched over to a linux build environment and all works perfekt - not deeper checked what is missing on windows side but code itselve is ok

knutwurst, 2021-03-08:
@m0rzan can you try to shorten the environment name and the build path? Are you using windows? If so, please try to download the latest 1.1.9 sources (attached to the release) and not the master branch, since this might be unstable.

Please close the issue, if this solves your problem :)

m0rzan, 2021-03-13:
Thank you for your answers. :)

Yes it seems to be a windows issue. Version 1.1.9 resulted in the same error as mentiond above. Shorten the filepath changed nothing. 

Building in a linux environment worked.

closed 2021-03-27 by knutwurst