knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #89: [FR] Minor change in BLtouch installation guide

gfornax, 2021-02-21:
Hi,

first of all thank you for the great, continuous work!

I have a small suggestion for the bltouch guides in the Wiki (github wiki unfortunately don't support pull requests).
Reason is I believe the necessity for setting the X/Y-Offests with M851 isn't stressed enough by mentioning it just at the bottom. Most people will most likely just follow the step-by-step list.
Therefore, I propose the following change:

18. M851 Z-0.6 (if your sensor is mounted at a different position than the FW's default, add your corresponding X and Y offset as well)

18. M851 Z-0.6 (falls der Sensor an einer anderen Position als dem FW-default angebracht ist, hier die zugehörigen X und Y-Offsets mitgeben)

Cheers

Andreas

hawkeyexp, 2021-02-21:
Hm, i think it's a classic rtfm thing :-) For my feeling the guide is clear.

Ich denke es ist absolut klar weil ja auch das Rechenbeispiel für den Offset mit drin ist :-)

gfornax, 2021-02-22:
Nachdem das Thema klar ist bleibe ich mal bei einer Sprache :)

Also mit RTFM hast du natürlich schon recht. Es wäre auch nicht dramatisch, wenn das nicht den Drucker zerstören könnte.

Wenn du die Guide von oben nach unten durcharbeitest wird dort M851 das erste mal bei Step 18 erwaehnt, Step 20 ist dann durchführen des levelings. Wenn der XY-Offset noch falsch (also off-bed, wie z.B. beim populären MK4) ist, fährt er dabei nach einiger Zeit voll gegen das Bett, die Z-Stop-Schalter werden ignoriert (oder ist das ein Bug?). So deutlich wie im Guide vor einem Verpolen des Sensors gewarnt wird fehlt im Gegensatz hier jegliche Warnung.

hawkeyexp, 2021-02-22:
Ich geh mal von mir aus aber da war es absolut logisch dass man den Offset beachten muss und beim ersten Test den Fingr am Abzug hat (Schalter) :-) Versteh es nicht falsch war nicht böse gemeint - aber ein Restrisiko bleibt bei jedem Eingriff in ein Gerät - das gilt ja hier genauso.

gfornax, 2021-02-22:
Hab das schon nicht falsch verstanden, keine Sorge, und ich habe durch genau diese Sicherheitsmaßnahme auch rechtzeitig abgeschaltet...
Denke aber trotz allem, das hier ein kleiner Hinweis einfach sinnvoll wäre.
Kostet nichts, macht die Doku nicht unübersichtlicher, verhindert aber ggf. den ein oder anderen materiellen Schaden und Frust.

knutwurst, 2021-03-27:
@gfornax ich schaue mir das mal an und übernehme das. Vielen Dank für den Hinweis! Es gibt ja immerhin so Einige, die gerade beim Lehmkühler X-Carriage Probleme mit dem Offset haben.

closed 2021-04-02 by github-actions[bot]