knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #90: [BUG] (short description) Red Light flashes during the print and now sensor doesn't work.

phyex0, 2021-02-28:
Dear developer, I was using the Bltouch sensor and your software for around 1 week. The sensor started to flash red light during the print. Since that moment I cannot use my sensor. I cannot level my printer or when I turn the printer on the sensor doesn't activate itself. It just flashes a red light. Connections are fine. What should I do?

phyex0, 2021-02-28:
https://www.reddit.com/r/AnycubicMegaS/comments/lued4i/guys_something_happend_during_the_print_i_started/

There is a video it'll make more sense

phyex0, 2021-03-06:
Dear developer are you there?

knutwurst, 2021-03-07:
@phyex0 , yes! I'm here! Sorry for the delay... 
Uhm.. as far as I can see in the /r, your sensor might be defective since the nozzle is not coming out and it fails the self test. Do you have a spare bltouch sensor to test?

phyex0, 2021-03-07:
What do you mean by spare? Any other sensor?

knutwurst, 2021-03-07:
Yes... another one you can test. For me it looks like a defective sensor.

phyex0, 2021-03-09:
I'll check all the connections via multimeter and then I'll try it :D I hope it's a connection issue. Thanks for the feedback bud.

closed 2021-04-15 by github-actions[bot]