knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #91: [BUG] (short description)

boredomwontgetus, 2021-03-05:
### Bug Description

When using VSCode to compile the MEGA_X_TMC Variant using the predefined Project Tasks the extruder fan does never start. 

### My Configurations

All on defaults from straight from the repo.

### Steps to Reproduce

1. Clone Repo
2. Open in VSCode
3. Choose Build on the MEGA_X_TMC Project Task
4. Flash firmware.hex to the printer

**Expected behavior:** 

The extruder fan to fire off autmatically as it does when using the MEGA_X_TMC precompiled firmware release

**Actual behavior:** 

Extruder fan never starts. Even when asked to start via octoprint.

#### Additional Information

knutwurst, 2021-03-05:
You're right! The Issue will be fixed asap!
In the meantime I would like to ask you to check out tag 1.1.9 or to load the sources from the 1.1.9 release. The current master is being converted for Chiron support and bugs (like this one) can occur. 

Thank you anyway for the hint!

closed 2021-03-05 by knutwurst

boredomwontgetus, 2021-03-05:
Thank you very much. And sorry for using saying "Extruder" what i meant was "Hotend"-Fan. I was in a bit of a hurry.