knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #92: [BUG] (short description)print will not resume after filament sensor pause.

paigewilliamson1, 2021-03-06:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Sn0wStorm, 2021-03-09:
Works fine for me, no problem. Just click on continue on the touch screen

paigewilliamson1, 2021-03-09:
hmm, it doesn't at all for me.

knutwurst, 2021-03-27:
What printer do you use? Can you make a video like this? https://www.youtube.com/watch?v=MRSr3vZjFYU

closed 2021-05-29 by github-actions[bot]