knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #94: [BUG] Linear-Advance and S-Curve acceleration at the same time?

ashleysommer, 2021-03-08:
I see in the README file (and in the marlin configuration files) that both S-Curve acceleration and Linear-Advance are enabled on this firmware.

I've been reading that those two settings don't really work well together:

See this Reddit discussion:
https://www.reddit.com/r/3Dprinting/comments/jipb8j/marlin_2_scurve_acceleration_linear_advance_and/
>"Linear advance is still incompatible with S curve acceleration. "

And this Marlin Bug report:
https://github.com/MarlinFirmware/Marlin/issues/14728

I've been doing some testing of my own, and so far I haven't actually found any problems when combining both features, but I haven't tested much.

ashleysommer, 2021-04-08:
Removing stale label.

knutwurst, 2021-04-23:
As long as you use LA and ScA on a cartesian printer, it works just fine. Only if you switch from classic jerk to junction deviation, JD won't work correctly.

closed 2021-05-15 by knutwurst