knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #95: [FR] Suggestion for "Setting up Cura" wiki page

ashleysommer, 2021-03-08:
I was eager to try out the out S-Curve acceleration when I installed this firmware on my Mega-S today.
But I didn't realize I wasn't actually using any of the new acceleration features because I had "enable acceleration control" and "enable jerk control" in Cura.

I don't know if acceleration control and jerk control are enabled by default in cura, or whether I had turned them on one time and forgot about them.

But there should be mention somewhere in the [wiki page](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/How-to-set-up-Cura) that Acceleration Control and Jerk Control settings should both be _disabled_ in Cura slicer settings if you want to take advantage of the fancy new Marlin algorithms and the fine tuned numbers baked into this firmware.

ashleysommer, 2021-04-08:
Removing stale label.

knutwurst, 2021-04-10:
I'm sorry I'm only now answering.
Acc and Jerk Control are normally deactivated in Cura and you have to activate them manually and set the values yourself. So it is more of a "pro feature", which is not always visible. I am of the opinion that someone only activates this slicer feature if they also know what they are doing.

No offense, but this is simply an operating error that I will not include in my documentation, as this is not a firmware feature, but a slicer configuration.

closed 2021-04-10 by knutwurst