knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #96: Anycubic mega pro tmc treiber

sims-60, 2021-03-08:
Hallo ich habe mir heute 3 tmc2208 treiber für meinen mega pro gekauft und ich wollte fragen ob wenn ich die jetzt einbaue ob ich dann die _tmc version nutzen kann weil ich baue die treiber ja nur für z und den extruder ein weil bei x und y ja schon tmc treiber sind wo die kabel ja schon umgedreht sind also will ich einfach nur fragen ob du das beachtest hast

hawkeyexp, 2021-03-08:
Hier musst du eins beachten:

bei der TMC-Version sind alle Drehrichtunen getauscht (was du ja mittels Kabel schon gemacht hast).
Also bleibt dir nur das eine oder anere:

- entweder die restlichen Kabel auch drehen und die normale Version weiter benutzen
- die bereits gedrehten Kabel wieder zurück drehen und die TMC-Version benutzen

Gruß

knutwurst, 2021-03-09:
Ich bin mir nicht sicher, ob ich das richtig verstanden habe. @sims-60 , hast DU selbst die Kabel gedreht, damit die Firmware läuft? Das musst du doch gar nicht, denn die MEGA_P Firmware beachtet den Original-Aufbau des Druckers, so dass X und Y schon in der Firmware invertiert sind.

Wenn du jetzt alle Treiber gegen TMC2208 tauschst, dann musst du die MEGA_P_TMC Firmware nehmen.

Es ist immer so gedacht:
 - Ohne TMC = Original (egal, welche Treiber vom Werk drin sind. Das ist alles schon bedacht)
 - Mit TMC = Alle Treiber sind TMC2208. Egal, ob da schon welche drin waren, oder eingebaut wurden.

sims-60, 2021-03-09:
Ne tut mir leid das ich das jetzt so sache auf x und y sind von werk aus c treiber

sims-60, 2021-03-09:
Mit dem c meinte ich tmc und bei x und y sind halt dann die kabel gedreht heißt jeder der ein mega pro hat das das Problem wie ich

knutwurst, 2021-03-09:
Du kannst eine Beträge auch editieren. Dann musst du sie nicht selbst beantworten ;) 

Und das höre ich zum ersten Mal. Und es gibt viele Mega Pro Nutzer. Beim Mega Pro muss man keine Kabel drehen und es sind auch vom Werk auch keine gedreht. Man muss nur die richtige Firmware installieren.

Welche hast du denn genommen?

sims-60, 2021-03-09:
Ich nutze gerade die standart version und wenn in ein paar tagen 3 tmc treiber kommen baue ich die in z und beim extruder ein

sims-60, 2021-03-09:
[image]
So sie das von werk aus heißt wenn ich tmc treiber einbaue muss NUR z und der extruder umgedreht werden

sims-60, 2021-03-09:
also wenn ich die 3 blauen treiber durch tmc ersetze kann ich dann die tmc firmware nehmen?

knutwurst, 2021-03-09:
Bist du dir sicher, dass das so gehört? Denn die Drehrichtung wird eigentlich in der Firmware vorgegeben. Bei dir sieht es so aus, als wenn die Stecker mal gedreht wurden, damit man eine MEGA S Firmware nutzen kann.

sims-60, 2021-03-09:
also ich habe nichts an den steckern gemacht und das war so von werk und ich nutze auch die mega pro firmware

knutwurst, 2021-03-09:
Aber du musstest nichts anpassen? Oder doch? 
Wenn jetzt die Knutwurst Mega Pro Firmware problemlos läuft, dann ist alles korrekt.

sims-60, 2021-03-09:
ok dann passt alles heißt wnn ich dann die 3 tmc treiber auch z und beim estruder einfach die _tmc version nehmen und es geht

knutwurst, 2021-03-09:
Genau :)

sims-60, 2021-03-09:
aber noch ein lob an dich xD so ein guten support gibt es nur hier

closed 2021-03-09 by sims-60