knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #97: [FR] MEATPACK enabled?

MrBill1234, 2021-03-08:
Is the MEATPACK pre-compiler option set in the pre-compiled binaries?

More info here:

https://github.com/scottmudge/OctoPrint-MeatPack/

MrBill1234, 2021-03-08:
Looking at the Marlin repository - it looks like it is only in the bugfix branch

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/meatpack.cpp

Could it be manually added and compiled in?

knutwurst, 2021-03-27:
At this point in time it is unfortunately not possible, as this feature is only available in the latest bugfix branch from Marlin, and I never update the entire branch, only individual elements. So I would have to extract the feature.

I don't know if I'll do that yet :) Maybe ... if there is a lot of time ...

:)

MrBill1234, 2021-03-27:
If it made its way into a main release branch, in theory it would be in your next release?

PR #185 (2021-08-20): Backport meatpack feature from upstream
Fixes #97 
Tested on MEGA-S (Old TFT, No BLT).
closed 2021-08-20 by knutwurst