knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #102: tmc2208 v1.2

sims-60, 2021-03-11:
hi ich habe mir jetzt die tmc 2208 treiber eingebaut aber habe jetzt das problem das ich kein  babystepping mehr habe habe das bett neu mit dem bl toch gelevelt aber ich fühle auch an meiner z achse keinerlei bewegungen aber wenn ich sie nen cm hoch oder runterfahren lassen funktioniert es hast du da eine lösung

sims-60, 2021-03-11:
mein  fehler funktioniert doch hatte nur nicht erwartet das es halt jetzt mit viel kleineren hochauflöserenden schritten es ausgleicht

closed 2021-03-27 by knutwurst

Marcel1337, 2021-05-12:
Hey @sims-60 ich wollte dich fragen, ob du nur die 3 neuen Treiber eingebaut hast (Für den Mega Pro) und von Knutwurst die TMC Version genommen hast?

sims-60, 2021-05-12:
ich habe TMC2208 von  FYSETC gekauft in dem fall in dem Princore shop weil man da die einzeln kaufen kann

sims-60, 2021-05-12:
@Marcel1337  hast du auch einen mega pro

Marcel1337, 2021-05-12:
Jo ich habe auch einen Mega Pro, hast du dann eine die _TMC Version von Knutwurst genommen, ohne etwas ändern zu müssen?

sims-60, 2021-05-12:
ja
wobei mitlerweile nutze ich klipper und habe mir da einen laser support reinprogrammiert