knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #103: [BUG] ABL not correct

serioustk, 2021-03-11:
### Bug Description
When running ABL, leveling is not done correct. Grid seems prone, test pattern does not look the same on the whole base.
Please find pictures below.

### My Configurations
MEGA_S_DGUS_TMC_BLT_10_v1.1.9.hex

### Steps to Reproduce

1. Install knutwurst marlin firmware
2. Install [original BLTouch](https://www.amazon.de/gp/product/B01FFV2TOS) according to [this](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(english)) tutorial.
2. Run auto bed leveling from special menu
3. Print [this](https://www.thingiverse.com/thing:2780118) leveling test [GCODE bed_leveling_test.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6124208/bed_leveling_test.zip)

**Expected behavior:** [What you expect to happen]
Test squares look identical in each corner of bed and in center.

**Actual behavior:** [What actually happens]
Test squares are not identcal. On the left side, test squares look fine while on the right side nozzle was too far away from bed while printing.

#### Additional Information
I use Anycubic i3 Mega S brought in early 2021 with blue/yellow display and mainboard without gorilla logo.
I experienced issues with an uneven ultrabase. Manually, I was able to level the corners correctly but the center was too low so I decided to add an ABL sensor.

If you look at the test print, the leveling map of the ABL seems to be prone to the right side. As you can see on the pictures, test squares on the left side (front and back) look as expected, center is not so well and the right side is far from fine.

What I have tried / checked so far:
- Reconfigured the ABL position as described in the tutorial several times on different spots on the ultrabase.
- Checked that the ultrabase is high enough so that the printer does not run into the Z stops while printing. In fact, it currently is about 4 mm higher than it should be with manual leveling (also tried even higher and lower).
- Changed the ultrabase height and run ABL again. Same results.
- Tried both `M420 S1` or `G29` after `G28` in gcode.
- Checked calibration of Z-motors. Both motors reach their Z stop at the exact same time and have the exact same distance from groundplate

I really can't figure why the test squares does not look identical. 
I'm glad to provide further information or do some testing if required.

#### Pictures
##### Overview
[image]
##### Front left
[image]
##### Center
[image]
##### Front right
[image]

v-a-d-e-r, 2021-03-12:
Same crap here with orig. BLTouch... :-(

[image]

adrioogle, 2021-03-12:
Same for me but I did not print these squares. But on the right side the nozzle is always way lower than on the left side.
I already printed some parts but I always put my parts on the front left and now the big part where is realized that.
I tried the same as you. Base higher and playing with Z offset but nothing worked.

[image]
[image]

v-a-d-e-r, 2021-03-12:
Ok, 3(+) people can not be stupid in the same way. ;-) I assume that the mesh values are not part of the Z-offset calculation (just ignored)!? I get by far more better print results with manual leveling and a piece of paper!!! That was not the goal of buying an (expensive) orig. BLTouch sensor....

haupthakan, 2021-03-12:
Yeah so I tried so things and got a bit better but it's still unusable.
Im not able zu use this printer till there is a solution.
I'll try some other firmwares maybe these will fix this issue.
If it fixes I'll post it in here

v-a-d-e-r, 2021-03-12:
Ok, this is my 2nd Anycubic Mega printer (the 1st one told me after two weeks the print bed has -14°C! I didn't believe that...*g* --> warranty case...). With manual leveling I get useable prints. But it is annoying to do leveling with a piece of paper. So I hope the issue will be fixed soon...

vergie, 2021-03-13:
Please check the correct vertical frame alligment on the left and right side (using a spirit level  could be enough). Then also check the vertical allignemnt of both X rods. 
If there is a differnce between the left and right side you have a problem with the BL Touch. Responsibe for the differenz in this case is the Y-offset of BL-Touch.  This Problem was described in detail here (in german) [https://makerware.thingiverse.com/thing:2230882]

haupthakan, 2021-03-13:
> Please check the correct vertical frame alligment on the left and right side (using a spirit level  could be enough). Then also check the vertical allignemnt of both X rods.
>
> If there is a [...]

I've read it and his idea is right
If your carriage would sag a bit the bl would sit lower so maybe my x axis is twisted.
But I've tried another firmware and it went better as with this one
Maybe I'll install the manual meshbed version for now and print the mk4 xcarriage where the y is almost 0 I think.
Thanks for your answer. I would've never think of this :)

haupthakan, 2021-03-13:
> Please check the correct vertical frame alligment on the left and right side (using a spirit level  could be enough). Then also check the vertical allignemnt of both X rods.
>
> If there is a [...]

I've checked the x rod alignment and they seam really the same so this doesn't seam to be the problem. :/ 
Even if there would be I wouldn't have found a way to adjust the x rods

haupthakan, 2021-03-13:
If someone is interested I'll put my mesh points in here. This is different almost everytime.
Between those I didn't touch anything. I just ran it couple of times

Recv:       0      1      2      3      4
Recv:  0 +0.565 +0.532 +0.490 +0.415 +0.357
Recv:  1 +0.500 +0.642 +0.622 +0.480 +0.387
Recv:  2 +0.578 +0.625 +0.598 +0.520 +0.395
Recv:  3 +0.612 +0.678 +0.617 +0.472 +0.420
Recv:  4 +0.675 +0.650 +0.573 +0.448 +0.307

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.347 +0.522 +0.482 +0.377 +0.312
Recv:  1 +0.522 +0.622 +0.622 +0.457 +0.360
Recv:  2 +0.600 +0.673 +0.607 +0.525 +0.425
Recv:  3 +0.710 +0.717 +0.610 +0.515 +0.420
Recv:  4 +0.740 +0.725 +0.605 +0.490 +0.327

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.382 +0.617 +0.558 +0.445 +0.365
Recv:  1 +0.652 +0.735 +0.698 +0.568 +0.430
Recv:  2 +0.710 +0.767 +0.707 +0.588 +0.462
Recv:  3 +0.770 +0.787 +0.707 +0.575 +0.477
Recv:  4 +0.835 +0.820 +0.673 +0.517 +0.325

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.565 +0.665 +0.630 +0.507 +0.418
Recv:  1 +0.698 +0.800 +0.757 +0.607 +0.475
Recv:  2 +0.773 +0.820 +0.740 +0.632 +0.495
Recv:  3 +0.830 +0.855 +0.755 +0.617 +0.510
Recv:  4 +0.905 +0.852 +0.715 +0.550 +0.372

v-a-d-e-r, 2021-03-13:
I also made 3 tests in a row, but the values are very close to each other....

< Bilinear Leveling Grid:
<       0      1      2      3      4
<  0 +0.987 +1.130 +1.231 +1.263 +1.231
<  1 +0.997 +1.113 +1.221 +1.256 +1.228
<  2 +0.955 +1.110 +1.253 +1.327 +1.347
<  3 +1.031 +1.142 +1.241 +1.285 +1.265
<  4 +1.066 +1.184 +1.263 +1.275 +1.251

< Bilinear Leveling Grid:
<       0      1      2      3      4
<  0 +0.985 +1.123 +1.226 +1.256 +1.231
<  1 +0.987 +1.103 +1.206 +1.251 +1.224
<  2 +0.945 +1.103 +1.251 +1.327 +1.354
<  3 +1.029 +1.137 +1.236 +1.283 +1.260
<  4 +1.059 +1.177 +1.258 +1.273 +1.253

< Bilinear Leveling Grid:
<       0      1      2      3      4
<  0 +0.980 +1.108 +1.214 +1.243 +1.219
<  1 +0.987 +1.100 +1.206 +1.248 +1.221
<  2 +0.940 +1.098 +1.241 +1.322 +1.354
<  3 +1.022 +1.132 +1.224 +1.268 +1.253
<  4 +1.054 +1.172 +1.248 +1.263 +1.236

haupthakan, 2021-03-13:
How did you get those values? Did you change something since your first post?
Are you still getting a bad first layer?
yeah I would be more than fine with these values :D but I'm not getting close to really good values :/
I haven't even really used my printer today because I'm not even close to a solution to this :( 
I've found a new mount for the BL-Touch where the probe is on the same Y as the nozzle.
Maybe this would help for the beginning.

v-a-d-e-r, 2021-03-14:
Nope, I changed nothing....

haupthakan, 2021-03-14:
I ran the probe accuracy test and the probe should be alright. I think it said something like 0.017. 
I'm really frustrated because manual mesh leveling works fine but it's very time consuming.

v-a-d-e-r, 2021-03-14:
Yeah, it is. But the print test I used is really extreme. It can also be that the temperature in the edges is too low.

serioustk, 2021-03-26:
> I've read it and his idea is right
> If your carriage would sag a bit the bl would sit lower so maybe my x axis is twisted.
> But I've tried another firmware and it went better as with this one
> Maybe [...]

This was the hint I needed, thank you!

For all who do not understand german here is a quick translation of the referenced post:
_The shafts of the X-Axis are not perfectly perpendicular to each other compared to the base. This varies depending on the X-position of the extruder unit. This leads to a different tilt of the extruder unit depending on the X-position. As the probe has a offset in Y direction compared to the extroder, the probe is sometimes more, sometimes less far away from the bed leading to wrong results._

I was able to adcust the X-shafts with a small spirit level loosing the holder screws of the shafts and turning them around a little. With a littie trying around, the spiril level now only shows a very small offset over the whole X-Axis leading to a very good looking test pattern using BLtouch calibration.

Thank you! This issue is solved and can now be closed.

closed 2021-03-26 by serioustk