knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #104: Disable Stealthchop mode for tmc2208 in the extruder

ar3dworld, 2021-03-15:
Hello,
I try to use the command m569 to disable Stealthchop mode for tmc2208 in the extruder but is returning and unknown command. with linear advance the tmc2208 for the extruder should not be in Stealthchop mode

v-a-d-e-r, 2021-03-15:
Here the same...
> M569
< echo:Unknown command: "M569"

Sn0wStorm, 2021-03-15:
The Mainboard does not have any UART Pins for the Stepperdrivers. This means the TMC2208 is running in standalone mode and settings can not be changed using firmware, including the mode.

I'm sure knutwurst can comment on this, but i think he patched the problem and the TMC2208 can be used in stealthchop for linear advance just fine.
For me linear advance works ok.

ar3dworld, 2021-03-15:
The problem is that the extruder is movinh alot from one side to the other when the linear advance is active. Its working but it seems too much small movements to the extruder. Can damadge the extruder? If i disable the linear advance the exteuder works well.

ar3dworld, 2021-03-15:
Can i use the tmc2208 for x y z and the a4988 for the extruder?

3dRikal, 2021-04-05:
> Can i use the tmc2208 for x y z and the a4988 for the extruder?

You can. I do at the moment. You will need to reverse the plug on the E0 but that's it.

closed 2021-05-15 by knutwurst