knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #106: BLTouch Smart v1.3 Not Working

CircuitGuy, 2021-03-21:
I have a v1.0 Trigorilla board on an Anycubic i3 Mega. TMC steppers (not relevant to this).

Wired everything up correctly (triple-checked!) for BLTouch v1.3.

The probe self-test passes, but when the probe triggers against the bed, Marlin takes no action.

What I've tried:
1. Check wiring. Twice.
2. In Configuratoin_adv.h uncommented the #define BLTOUCH_SET_5V_MODE, and build with the MEGA_S_TMC_BLT_10 PlatformIO config.

Once I set 5V mode, the BLTouch won't extend pins under manual control. Feel like maybe I'm making a noob mistake...

After a lot of googling, I did find an identical issue in one of the davidramiro forks.
https://github.com/MNieddu91/Marlin-Ai3M-2.0.x-BLTouch/issues/3

agentchimendez, 2021-03-21:
Mine ended up being a wiring issue. I’m really bad at soldering and I’m new to multimeters. What finally clued me in was a continuity test with the tone. 

Turned out I had the white wire connected but not the black I believe. Rather than attempting to solder again I skipped the board and spliced directly into the bundle of wires above the green clip connector.

CircuitGuy, 2021-03-25:
@agentchimendez Thanks. I continuity checked everything just now. Still not working.

 This thing gets *really* hot just sitting there, not even with the sensor wire plugged in, just 5.03 V and the servo connection. It works at first, i.e. will pass the power-on test and respond to probe commands. If I let it self-heat for too long, eventually it stops responding to the servo commands around 80 *C according to my thermal camera. Looks like it's damaged. Going to assume it's my fault somehow. :( Probably buying a clone and trying again.

closed 2021-03-25 by CircuitGuy