knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #107: TMC2208 weird layers MEGA X

timoth31590, 2021-03-21:
Hello,

With the TMC2208, I'm getting bad layers, they're getting over the edge like the bottom cube here https://preview.redd.it/9rlnz9kmx1o61.jpg?width=640&crop=smart&auto=webp&s=b67e2bc3c3aed329deda24061939db3e27a70bd2 . I have had a really long help on reddit, seems to come from acceleration, but even with your setting I'm getting this problem. I'm using a Dragonfly hotend and the E3D Titan Extruder, all my settings for those two are checked and okay

```
echo:; Maximum feedrates (units/s):
echo:  M203 X120.00 Y120.00 Z18.00 E80.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X400.00 Y400.00 Z60.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P400.00 R3000.00 T400.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X4.00 Y4.00 Z0.20 E8.00
```

knutwurst, 2021-04-21:
Hi, sorry for answering so late.

To me it looks like your extruder is not feeding properly. Which extruder did you use and which reference voltage does the driver of E0 have? I think that is where the problem lies.

timoth31590, 2021-04-21:
Hello, I have a E3D Titan with the Compact but powerful motor, steps are calibrated. It works fine with petg and the same layer heigh but with PLA I get those weird layers. I've swapped the 2208 for a 2209 and put it at 1V, it looks better but I still get some layers going over the edge

closed 2021-05-15 by knutwurst