knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #108: [BUG] MEGA_X_v1.1.9.hex reports incorrect machine_type

chiccorosso, 2021-03-23:
### Bug Description
I just flashed my Anycubic Mega X and it reports wrong machine type:
```
Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.5.4 (knutwurst, Github) KNUTWURST_AI3M_VERSION:1.1.9 DISTRIBUTION_DATE:2021-01-13 SOURCE_CODE_URL:https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic i3 MEGA EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d
```

### My Configurations

Mega X out of the box

### Steps to Reproduce

1. update firmware via octoprint plugin
2. type M115 on terminal

**Expected behavior:** [What you expect to happen]
It should report "Knutwurst's Anycubic MEGA X"

knutwurst, 2021-03-23:
That's not a bug. It's the way, this firmware is built. It uses the same codebase for all supported printers and the machine type string is hardcoded and gereric.

"i3" stands for the Prusa i3 Clone and "MEGA" for the cartesian printer family from Anycubic. That's all your Mega X is like :) I might change the string to a more generic one in the future, when the Chiron is fully supported.

closed 2021-03-23 by knutwurst

chiccorosso, 2021-03-23:
Thank you for the quick answer!