knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #113: [BUG] Using the Touch Screen Resets the printer

rianweaver, 2021-03-30:
### Bug Description

I've found that using the touch screen on my Mega X after being flashed with Knutwurst often resets the printer.  This will mess up running prints when trying to do a filament change, etc. 

### My Configurations

**Required:** used downloaded compiled firmware

### Steps to Reproduce

1. Using the touch screen, hit buttons (like filament change), more than once, in quick succession
This happens intermittently, both during a print, and during "idle"

**Expected behavior:** On screen buttons work as expected

**Actual behavior:** Intermittently, it sends a reset to the printer.  Any expected temperatures are reset to zero, printer beeps, and resets.  Reset detected in Octoprint.   

#### Additional Information

rianweaver, 2021-03-30:
Here are my octoprint logs if that's helpful.    It seems maybe the M110 is what's being sent by the button presses?

2021-03-29 21:39:09,046 - octoprint.util.comm - WARNING - Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
2021-03-29 21:39:09,051 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-29 21:39:09,061 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Cancelling"
2021-03-29 21:39:09,063 - octoprint.util.comm - INFO - Force-sending M108 to the printer
2021-03-29 21:39:09,115 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: MrsWeaverNameplate.gcode, owner: rian.weaver, user: None
2021-03-29 21:39:09,194 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2021-03-29 21:39:09,246 - octoprint.plugins.filamentreload - INFO - PrintCancelled: Disabling filament sensor.
2021-03-29 21:39:09,248 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-03-29 21:39:09,430 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-29 21:39:09,558 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2021-03-29 21:39:09,564 - octoprint.plugins.filamentreload - INFO - PrintFailed: Disabling filament sensor.
2021-03-29 21:39:09,884 - octoprint.plugins.tracking - INFO - Sent tracking event print_cancelled, payload: {'origin': 'local', 'file': 'fd2a8d9d6565b80c188aee8ae4dc4e414b680f12', 'elapsed': 10913}
2021-03-29 21:39:10,516 - octoprint.plugins.tracking - INFO - Sent tracking event print_failed, payload: {'origin': 'local', 'file': 'fd2a8d9d6565b80c188aee8ae4dc4e414b680f12', 'elapsed': 10913, 'reason': 'cancelled'}
2021-03-29 21:39:22,255 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"
2021-03-29 21:39:22,262 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-state' with value Operational (Mega X)

3dRikal, 2021-04-05:
Actually I think this is related to the filament detection issue. I do get the same kind of problem but only when the filament is removed from the printer. Only way to get the printer to respond again is to reset.

knutwurst, 2021-04-21:
This is a known issue with the first Mega X with original DGUS II Display (see [here](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Types-of-Anycubic-Touchscreens)) which causes the Display to lock up and reset. Also in v1.2.0 the special menu handling is optimized, which should help to avoid the issue, but I can not say it for sure :/

closed 2021-05-15 by knutwurst