knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #115: MEGA_S_DGUS_TMC_BLT_10_v1.x.x Versionen brechen in OctoPrint mit Kommunikationsproblemen ab

aguthard, 2021-03-31:
Die Versionen MEGA_S_DGUS_TMC_BLT_10_v1.1.9 und MEGA_S_DGUS_TMC_BLT_10_v1.2.0-Alpha.1 haben in Verbindung mit OctoPrint auf einem Raspberry Pi 3B direkt nach dem Auslesen der Gerätedaten Kommunikationsprobleme, die dann zum Abbruch der Verbindung führen.

Alternativ wurden folgende Versionen getestet: MEGA_S_TMC_BLT_10_v1.1.9 und MEGA_S_DGUS_TMC_10_v1.1.9
Diese funktionieren beide einwandfrei, sodass ich einen Fehler in der Grundkonfiguration (Verkabelung etc.) ausschließen würde.

sims-60, 2021-03-31:
hi hast du ein display wo der hintergrund weiß und die symbole weiß sind oder hast du ein display wo die schrift gelb und der hintegrund blau ist und hast du einen bl touch installiert?

aguthard, 2021-04-01:
Mein Display ist die blau/gelbe Variante, ein BL Touch ist installiert.

Beide Komponenten funktionieren mit anderen Firmwarevarianten einwandfrei.

Einzig die Konstellation blau/gelbes DGUS Display UND TMC Treiber UND BLT scheint ein Problem zu erzeugen.

In Issue #88 ist von Kompilierungsproblemen von genau dieser Konstellation die Rede.
Könnte das ein Ansatzpunkt sein?
 
Getestet habe ich mit  fertig kompilierten .hex Dateien aus dem GIT.
Selbst kompilieren scheiterte auf meinem Windows 10 Rechner reproduzierbar an den in Issue #88 geschilderten Problemen.

aguthard, 2021-04-01:
[serial.log](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6241183/serial.log)

Anscheinend ist das Problem im Bereich des Zugriffs auf die SD Karte eingrenzbar (siehe Log).
Die SD Karte ist im Gerät eingesteckt.

knutwurst, 2021-04-10:
Hast du mal versucht, die SD Karte raus zu lassen?

aguthard, 2021-04-11:
OHNE SD Karte funktionieren alle getesteten Varianten, inklusive der vorher auffälligen DGUS_TMC_BLT Kombinationen (getestet mit V1.1.9, V1.2.0-Alpha1 und 1.2.0-Alpha2)

Für mich ist das so in Ordnung.
Damit kann ich leben.

Die Frage bleibt, warum sich bei der DGUS_TMC_BLT Kombination in Verbindung mit einer gesteckten SD Karte die Kommunikation aufhängt, im Vergleich zu den "kleineren" Kombinationen.

@knutwurst:
Vielen Dank für deine bis hierher geleistete Arbeit und den im akuten Fall entscheidenden Tipp!
Trivial, aber sehr effektiv! :-)

knutwurst, 2021-04-11:
Ich kann es mir nur so erklären, dass über USB mehr RAM verwendet wird und gerade wenn weitere Features aktiv sind, bleibt nicht viel übrig. Besonders die DGUS Variante arbeitet hart am Limit. Ich habe allerdings schon einen Fix dafür, welcher in die neue Version kommt.

Schön, dass es so klappt. :)

closed 2021-04-11 by knutwurst