knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #116: MeshLeveling MEGA_S_TMC_BLT_10_v1.2.0-Alpha.1

KuhnDaniel, 2021-04-05:
### Bug Description

After FW update, the Mesh start at maximum of 3 Points, then STOP

I was updated 3 times, same result

KuhnDaniel, 2021-04-05:
Update: 
If the command starts with Pronterface, it does that!
NOT from the TFT menu

KuhnDaniel, 2021-04-06:
In the monitor:
Special Menu: Preheat Ultrabase
ok
Special Menu: BLTouch Leveling
echo:busy: processing
echo:busy: processing
echo:busy: processing
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
ok
echo:busy: processing
Error:!! STOP called because of BLTouch error - restart with M999
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
echo:busy: processing
echo:busy: processing
Error:Probing Failed
X:0.00 Y:27.00 Z:11.66 E:0.00 Count X:0 Y:2160 Z:8666
ok
echo:endstops hit:  P:1.66
echo:Settings Stored (712 bytes; crc 9271)
ok

hawkeyexp, 2021-04-06:
is your pin releasing before the error or is a missing pin release your problem which triggers the fail?

if the pin release is the problem have a look here: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/98

KuhnDaniel, 2021-04-06:
Der Prozess startet wie gewohnt, Sensor fährt aus, Z fährt runter, löst aus , Z geht wieder hoch, die LED Blinkt wie gewohnt!
Manchmal macht er 1 Messung, manchmal 3. Mehr Messungen gab es noch nie!
Beantwortet das dien Anfrage?

The process starts as usual, the sensor extends, Z moves down, triggers, Z goes up again, the LED flashes as usual!
Sometimes he takes 1 measurement, sometimes 3. There have never been more measurements!
Does that answer your request?

KuhnDaniel, 2021-04-06:
Wenn ich die FW 1.9.x flashe, läuft alles ohne Probleme. 
Ich habe den Eindruck, es könnte mit der Neopixel Geschichte zu tun haben. 
Falls da D11/D5 (je nach Board Trigorilla / 1.1 / 0.0.2) das Problem macht, seit der neuen Marlin 2.x!
Ich habe gerade die beiden Versionen runtergeladen und schau mir das an.

Es kann kein technisches Problem sein, es muss in der Config stecken.

When I flash the FW 1.9.x, everything runs without problems.
I got the impression it might have something to do with the Neopixel story.
If the D11 / D5 (depending on the board Trigorilla / 1.1 / 0.0.2) causes the problem, since the new Marlin 2.x!
I just downloaded the two versions and am looking at this.

It can't be a technical problem, it has to be in the config.

knutwurst, 2021-04-10:
@KuhnDaniel , thank you for the information! I'll dig into it.

KuhnDaniel, 2021-04-10:
> 
> 
> @KuhnDaniel , thank you for the information! I'll dig into it.

Danke Oliver, für die Rückmeldung.
Mit der 1.1.9 läuft alles wie gewohnt. Hardwarefehler sollte ausgeschlossen sein, m.M. nach

KuhnDaniel, 2021-04-11:
> 
> 
> @KuhnDaniel , thank you for the information! I'll dig into it.

Hallo Oliver,
mache mir gerade Gedanken und habe folgende Feststellung zum Thema AuoMesh!
Mein Z-Offset liegt bei 1.70 ( 0.06 - 1.76) beim letzten Messen.
Nun zeigt ja der Probe Fehler bei 1.66 (endstops hit: P:1.66)
Wäre ja, in der Logik ein Minus Wert unter dem Z-Stopp!
Denke/Verstehe ich das richtig?!

Danke für dein Feedback.
Gruss aus Basel, Daniel

frobert, 2021-04-17:
Check bltouch cable connections (black and white), had the same issue, was a bad cable connection.

KuhnDaniel, 2021-04-17:
> 
> 
> Check bltouch cable connections (black and white), had the same issue, was a bad cable connection.

thanks robert, the cables are all ok. With 1.1.9 everything works, even if I go back from 2.0.x - where the error is - the AutoMashLeveln simply works again.

knutwurst, 2021-04-21:
There was a "bug", which caused several non-genuine BLTouch Sensors (like the 3DTouch) to fail with Firmware 1.2.0. The bug has been fixed in 1.2.0-beta.2.

closed 2021-04-21 by knutwurst