knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #117: Cannot see special menu on Mega X with "new" blue / yellow DGUS II display

gitshanghai, 2021-04-06:
New Mega X cannot see special menu

_DGUS stands for the "new" blue / yellow DGUS II display, which otherwise cannot show a special menu. (Only relevant for MEGA S ?)

gitshanghai, 2021-04-06:
I flashed MEGA_S_DGUS_v1.1.9.hex and I can see the special menu! So please generate an updated hex file for Mega X , thanks

knutwurst, 2021-04-06:
The _DGUS Extension is only relevant for the Mega S. You just need to flash the MEGA_X version.

gitshanghai, 2021-04-06:
Thanks for your fast reply,  for my Mega X I tried:
1.  Flashed the MEGA_X -> NO Menu
2. Flashed the S Version just to try to see the Menu -> WITH Menu 

So there must be an issue with my Display hardware on my MEGA X.

knutwurst, 2021-04-06:
Okay? That's weird... Did you use the DGUS Version for the MEGA S?
So maybe Anycubic updated the TFT Firmware. Can you take some photos of the Display itself from the inside? Or can you tell me which Display you have when you look at this? -> https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Types-of-Anycubic-Touchscreens

gitshanghai, 2021-04-06:
Yes, used MEGA_S_DGUS_v1.1.9.hex Pictures will send you tomorrow.

gitshanghai, 2021-04-07:
Definitely the new one, please also see pictures
[image]
[image]

Comparison of Anycubic Touchscreens:
**Right: The "new" Mega S and Mega P TFT**
 https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Types-of-Anycubic-Touchscreens

gitshanghai, 2021-04-08:
@knutwurst can you please provide an updated firmware MEGA_X_v1.2.0-Alpha.1.hex with DGUS support, thanks

knutwurst, 2021-04-10:
I see ... seems like anycubic now uses the Mega P Display on all printers :/ 
I'll update the Code in the next alpha version.

gitshanghai, 2021-04-10:
Thank you,  looking forward to try

knutwurst, 2021-04-10:
Fixed in bc46dcb. 
Will be released in 1.2.0-alpha.2.

closed 2021-04-10 by knutwurst