knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #119: [BUG] Time since start printing is not displayed - 1.2.0-alpha.1

danzbox, 2021-04-08:
### Bug Description
When using pre-compiled version of 1.2.0-alpha.1 for Mega M 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.2.0-alpha.1/MEGA_v1.2.0-Alpha.1.hex
the time in the display is not displayed correctly during print. While it updates every minute on 1.1.9 I have not found what it displays on the new release, but it looks to me it is the time when the print was start (total time since turned on) and it does not get updated.

### My Configurations

Anycubic i3 Mega (M)
MEGA_v1.2.0-Alpha.1.hex
Print from SD Card

### Steps to Reproduce

1. Insert SD Card into printer
2. Turn on the printer
3. Preheat to PLA
3. Select a Gcode-file from SD to print
4. Observe display

**Expected behavior:** 
Time since print started shown in minutes on display and updated every minute (increment by one minute)
Total time printed shown when print done message appears

**Actual behavior:** 
Probably time since printer is turned on shown in display static
--:-- shown when print done

#### Additional Information

[image]

knutwurst, 2021-04-10:
Thank you for the helpful information. It's strange, since not all user have these issues. but I'll dig into it!

knutwurst, 2021-04-10:
Fixed in 7cdc56c and will be available in the next release (1.2.0-alpha.3 I think..)

closed 2021-04-10 by knutwurst

danzbox, 2021-04-12:
Thank you for looking into it.