knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #120: weird sound of the Z-as motors.

projectwind, 2021-04-12:
Hello,
Z-axis motors making a lot more noise when moving
X and Y motors sounds perfect 
i use the TMC2209
MEGA_X_TMC_BLT_10_v1.2.0-Alpha.1.hex

Can you send me the right Hex file? for The 
mega X TMC2209 with BLT?
ramon_wind@hotmail.com

thanks a lot 
Regards

Ramon

knutwurst, 2021-04-12:
Hi Ramon,
There is no version specifically for TMC2209, as these work exactly like TMC2208. If the motors are very loud, either the reference voltage is incorrect or the acceleration values are too high. Have you loaded the default settings? Which VRef do you use? Did you calculate it correctly? See: https://wiki.fysetc.com/Silent2209/

projectwind, 2021-04-13:
thanks Knutwurst for your fast reply.
oke we flashed the firmware to PreRelease is 1.2.0-alpha.3
Yes we loaded the default settings
we use the VRef 0.95
but i don't know what the RMS Current (A) to calculate the Reference Voltage (V).
do you know the  RMS Current (A) from PreRelease is 1.2.0-alpha.3?

Thanks for you help

Ramon Wind

knutwurst, 2021-04-13:
The max current of your stepper motors is 900mA. To calculate the RMS, just do it like this:  900mA peak = 0.9/1.414 = 0.636mA Irms. - So 0.95 Vref should be fine. This has nothing to do with the firmware! It's the current you provide to your stepper motors. And the reference voltage is only a "reference" so you don't need to measure the current. The reference voltage is just another representation of the maximum  current, the stepper driver provides to the motor. So you need to find the sweet spot between "stepper driver too hot" and "motor too weak". Sometimes you need more power, when there is a quick direction change. So on Y I would recommend not going under 1 Vref.

knutwurst, 2021-05-21:
Hi @projectwind , is this still an issue?

projectwind, 2021-05-21:
Hello, Yes X and Y are perfect
Z as makes noise
But it works.

Thanks for your question,

Regards

Ramon

Verstuurd vanaf mijn iPhone

knutwurst, 2021-05-21:
Great :)
So I'm going to close this ticket. 👍🏻

closed 2021-05-21 by knutwurst