knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #122: Fan Speed while printing

gloomytree, 2021-04-17:
Hi, 

i have found a minor issue. 

While printing, the fan speed is not displayed correctly in the menu where you can change it. Once you set a manual value, e.g. 80%, it is displayed correctly. I set the fan speed to 75% in the slicer, the display shows 0%.

While printig:
 - back
 - speed
 - fan speed

I am using MEGA_S_TMC_BLT_11_v1.1.9.hex on an Anycubic I3 Mega S.

Ty for all the work!

knutwurst, 2021-04-21:
Thanks for the info! I'll take a look into that.

knutwurst, 2021-05-15:
Does the issue still persist on the 1.2.0 Version? I just tried and it works just fine, showing 70% (form gcode) and I could change the speed to whatever I want.

gloomytree, 2021-05-31:
I've not jet tested 1.2.0, but I wanted to let you know that I will flash it and test when I have some spare time to tinker with the printer!

knutwurst, 2021-06-29:
The 1.2.0 final is out. Did you test it?

knutwurst, 2021-07-30:
@gloomytree is the issue fixed? Can the ticket be closed?

closed 2021-08-05 by github-actions[bot]

gloomytree, 2021-08-06:
Hi, I tried to test this today, but after flashing the new firmware version the homing went right into the bed. Dunno what happend, have to debug it now...

gloomytree, 2021-08-08:
Ok, the BLTouch is working again, it was no problem with the firmware. Also I can confirm that the fan speed is displayed correctly on the first try. 

Btw. the new version is awesome at the first glance. BLTouch leveling is so much faster... TY again =)