knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #124: [BUG] error building MEGA_S_DGUS

aginies, 2021-04-18:
### Bug Description

Just after cloning i can not compile the MEGA_S_DGUS using the project.

### My Configurations

Nothing specific, everything by default.

### Steps to Reproduce

1. clone repo
2. go to project MEGA_S_DGUS
3. clic build

Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/dogm/u8g_fontutf8.cpp.o
Marlin/src/lcd/HardwareSerial.cpp: In function 'void __vector_54()':
Marlin/src/lcd/HardwareSerial.cpp:92:19: warning: unused variable 'c' [-Wunused-variable]
     unsigned char c = UDR3;
                   ^
Marlin/src/lcd/HardwareSerial.cpp: In member function 'void HardwareSerialClass::begin(long unsigned int, byte)':
Marlin/src/lcd/HardwareSerial.cpp:185:11: warning: unused variable 'current_config' [-Wunused-variable]
   uint8_t current_config;
           ^~~~~~~~~~~~~~
Marlin/src/lcd/HardwareSerial.cpp: In member function 'virtual size_t HardwareSerialClass::write(uint8_t)':
Marlin/src/lcd/HardwareSerial.cpp:292:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   while (i == _tx_buffer->tail)
          ~~^~~~~~~~~~~~~~~~~~~
Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/dogm/ultralcd_DOGM.cpp.o
Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp.o
Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/extui/lib/dgus/DGUSDisplay.cpp.o
Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/extui/lib/dgus/fysetc/DGUSDisplayDef.cpp.o
Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/extui/lib/dgus/hiprecy/DGUSDisplayDef.cpp.o
Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/extui/lib/dgus/origin/DGUSDisplayDef.cpp.o
Marlin/src/lcd/anycubic_touchscreen.cpp: In function 'void setupMyZoffset()':
Marlin/src/lcd/anycubic_touchscreen.cpp:70:20: error: 'NOZZLE_TO_PROBE_OFFSET' was not declared in this scope
     probe.offset = NOZZLE_TO_PROBE_OFFSET;
                    ^~~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/anycubic_touchscreen.cpp:70:20: note: suggested alternative: 'STR_PROBE_OFFSET'
     probe.offset = NOZZLE_TO_PROBE_OFFSET;
                    ^~~~~~~~~~~~~~~~~~~~~~
                    STR_PROBE_OFFSET
Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/extui/lib/ftdi_eve_touch_ui/archim2-flash/flash_storage.cpp.o
Compiling .pio/build/MEGA_S_DGUS/src/src/lcd/extui/lib/ftdi_eve_touch_ui/archim2-flash/media_file_reader.cpp.o
*** [.pio/build/MEGA_S_DGUS/src/src/lcd/anycubic_touchscreen.cpp.o] Error 1

knutwurst, 2021-04-19:
The build bug in the baster branch has been fixed. But in general it would make sense if you don't take the source code from the master, but from one of the releases. It's there as a ZIP in the assets. Or you check out the last release tag.

aginies, 2021-04-19:
Thx for the reply. I switched to a stable release and this is buillding fine. As you were in alpha in the master branch i reported the bug, but probably you were alaready aware about it as this is a devel release.

closed 2021-04-21 by knutwurst