knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #125: [BUG] error building MEGA_X_DGUS_TMC

SpRiNgToIfEl, 2021-04-18:
### Bug Description

compile the firmware for mega x with enabled *BMG*.
i add in the i3_mega_envs.ini @ MEGA_X_DGUS_TMC
-DKNUTWURST_BMG 

### My Configurations

Nothing specific, everything by default. only enable BMG

### Steps to Reproduce

1. download the alpha 3 zip, extract, open it in VSC and platformio
2. edit  i3_mega_envs.ini, save
3. @ platfornio > project > MEGA_X_DGUS_TMC > build

Compiling .pio\build\MEGA_X_DGUS_TMC\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\text_box.cpp.o
Compiling .pio\build\MEGA_X_DGUS_TMC\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\tiny_timer.cpp.o
Compiling .pio\build\MEGA_X_DGUS_TMC\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_bitmaps.cpp.o
In file included from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./AVR/HAL.h:31:0,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/MarlinConfig.h:30,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../ui_api.h:45,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../compat.h:31,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../config.h:24,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../compat.h:20,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../ftdi_extended.h:25,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_bitmaps.cpp:22:
Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./AVR/MarlinSerial.h:36:39: fatal error: ../../inc/MarlinConfigPre.h: No such file or directory
compilation terminated.
*** [.pio\build\MEGA_X_DGUS_TMC\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_bitmaps.cpp.o] Error 1
======================================================================================================================================== [FAILED] Took 116.59 seconds ========================================================================================================================================
Environment      Status    Duration
---------------  --------  ------------
MEGA_X_DGUS_TMC  FAILED    00:01:56.592
=================================================================================================================================== 1 failed, 0 succeeded in 00:01:56.592 =================================================================================================================================== 
Der Terminalprozess "C:\Users\STEFAN\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'MEGA_X_DGUS_TMC'" wurde mit folgendem Exitcode beendet: 1.

SpRiNgToIfEl, 2021-04-18:
[image]

SpRiNgToIfEl, 2021-04-18:
same issue with version 1.2.0 alpha v0 - v3.

compile the firmware with version 1.1.9 works fine.

knutwurst, 2021-04-19:
Hi,
The path with the backslashes looks like Windows. Unfortunately, Windows has problems with paths that are too long, so some builds are working but others don't.

It looks like a dependency is not being downloaded correctly or is no longer found. Sometimes it helps to just start the build process again and sometimes it helps to keep the path as short as possible and maybe also to shorten the name of the project.

However, yesterday I also changed something in the source code and fixed some bugs. As you can see (also in the GitHub CI), the current master builds through without any problems.

Nevertheless, I recommend always using the latest release and not the master, as the code can be unstable.

It's also always a good idea to see if the source code has a green tick or a red X if you really want to build from the master.

SpRiNgToIfEl, 2021-04-20:
hi, i tried again and took the same path length. unfortunately it is not possible to compile the firmware.

closed 2021-05-20 by knutwurst