knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #127: MEGA S TMC BLTOUCH DOESN'T WORK

wuyonghuahk, 2021-04-19:
When the bltouch needle touchs the bed, Z axis don't stop at all. It can work well by 1.1.9 firmware.I test serival times,but it steel dosn't work.

knutwurst, 2021-04-21:
Which firmware version do you use? Which one works, which doesn't?

wuyonghuahk, 2021-04-21:
Now,I use 1.1.9.     2.0 doesn't work.

knutwurst, 2021-04-21:
1.2.0 Alpha or beta?

Thank you for your input! I'll take a look into that.

wuyonghuahk, 2021-04-21:
1.2.0 Alpha

knutwurst, 2021-04-21:
Does the problem sill persist in 1.2.0-beta.1?

wuyonghuahk, 2021-04-21:
yes, bltouch doesn't work.

knutwurst, 2021-04-21:
Ok, thank you very much! I'll look into that!

wuyonghuahk, 2021-04-21:
Not at all. I hope you can fix the issue.

knutwurst, 2021-04-21:
Can you tell me which version you used exactly? The complete filename helps a lot.

wuyonghuahk, 2021-04-21:
I had used the two versions.

wuyonghuahk, 2021-04-21:
I use TMC2209.

knutwurst, 2021-04-21:
@wuyonghuahk can you please tell me the filename of the firmware you used?

wuyonghuahk, 2021-04-21:
MEGA_S_TMC_BLT_10_v1.2.0-Alpha.3 and MEGA_S_TMC_BLT_10_v1.2.0-beta.1

knutwurst, 2021-04-21:
Please try the attached hotfix firmware it this solves your issue :) 
[MEGA_S_TMC_BLT_10_v1.2.0-b1hotfix1.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6349835/MEGA_S_TMC_BLT_10_v1.2.0-b1hotfix1.hex.zip)

wuyonghuahk, 2021-04-21:
ok, I'll try.

wuyonghuahk, 2021-04-21:
BLTOUCH doesn‘t work.

wuyonghuahk, 2021-04-21:
I made a video, but I couldn't send it by email.

knutwurst, 2021-04-21:
That's because we are not communicating via email. It's the github Issue here: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/127

You can upload zip files if you want, but I thing I don't need a video.

Here is a new possible fix. Please try it out: 
[MEGA_S_TMC_BLT_10_v1.2.0-b1hotfix2.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6350155/MEGA_S_TMC_BLT_10_v1.2.0-b1hotfix2.hex.zip)

wuyonghuahk, 2021-04-21:
It works,thank you very much!

wuyonghuahk, 2021-04-21:
what's the problem？

knutwurst, 2021-04-21:
It seems like the the software-mode for the Sensor does not work on all (older) models. Original BLTouch 3.1 works fine, but there are some issues when it comes to pulse lengths on other BLTouch models or clones.

I'll merge the fix into the official sources now and build a 1.2.0-beta.2.

Thank you very much for your help and the fast replies! 

Have an nice day <3

closed 2021-04-21 by knutwurst

wuyonghuahk, 2021-04-21:
I use the 3D touch. Bltouch is expensive.

knutwurst, 2021-04-21:
Yes, that was my thought :) 
The first beta only works with original BLTouch v3.

wuyonghuahk, 2021-04-21:
got it.