knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #128: G29 probes ones and stops

joloman2, 2021-04-19:
Hello i have a anycubic mega x with bltouch 

if i send a g29 command from octoprint it probes ones and the it stops i use MEGA_X_DGUS_TMC_BLT_10_v1.2.0-beta.1.hex firmware it gives me this error WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Probing Failed 

can you help me regards 

Joloman2

wes1993, 2021-04-19:
Hello,
have you seen this?

Troubleshooting
If everything appears correct and the automatic leveling starts, but stops at the second measuring point and the error message "Probing Failed" appears in Pronterface, this is an indication that the Z-offset was not set correctly and the sensor suddenly reaches a point which is below the height of the nozzle.

joloman2, 2021-04-20:
oke how do i set the z offzet

wes1993, 2021-04-20:
First you need to check if the BLTouch distance from bed is correct, could you upload a photo of your BLTouch installation with the BLTouch push pin down?

joloman2, 2021-04-20:
[image]

joloman2, 2021-04-20:
[image]

wes1993, 2021-04-20:
I can’t see very well but probably you need to add a thickness from sensor and the cage I have also do this because the sensor stay high.. no I can’t do a photo of mine because it’s printing..

joloman2, 2021-04-20:
oke the probe is now lower than the nozzle how do i calculate the z offset

wes1993, 2021-04-20:
Now you need to repeat this steps here:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(english)

knutwurst, 2021-04-21:
The current alpha/beta releases for 1.2.0 are kinda buggy when it comes to BLTouch :/ I'm sorry for that.

Please try this hotfix version: 
[MEGA_X_DGUS_BLT_10_v1.2.0-b1hotfix1.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6349548/MEGA_X_DGUS_BLT_10_v1.2.0-b1hotfix1.hex.zip)

joloman2, 2021-04-22:
thanks that fixed it

knutwurst, 2021-04-22:
Great! 
Let me guess: you're using a non-genuine BLTouch like the 3DTouch, right?

joloman2, 2021-04-22:
nope im using a genuine bltouch

closed 2021-05-20 by knutwurst