knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #129: [BUG] Just an help

wes1993, 2021-04-19:
Hello,
i have everything configured and works well, but i have one question, it's normal that when i do this step:
Place a sheet of paper under the nozzle
Slowly lower the nozzle with G1 Z-1 or G1 Z-0.1 or G1 Z-0.02 until the paper is difficult to move.

My printer reach the endstop? so i have this result:
X:38.00 Y:15.00 Z:0.00 E:0.00 Count X:3040 Y:1200 Z:0

And also the paper it's almost free, i can't reach the point that the paper is difficult to move.

I have executed a manual level before starting this procedure.

Best Regards
Stefano

knutwurst, 2021-04-21:
If the paper is almost free, your previous manual leveling was not correct. You need to level to z=0. So the nozzle almost touches the bed. Not 0.02, not 0.02 ... 0. It is needed for the BLTouch sensor to know where the bed position is.

wes1993, 2021-04-21:
Thanks for your reply :)
I have used the menu manual levelling option that are point A B C D and I have levelled with the paper but when I do with BLTouch procedure the end stop arrive

knutwurst, 2021-04-22:
That can't be physically possible. If you have leveled correctly, the nozzle should touch the bed when the end stops trigger. They cannot trigger beforehand, because that would mean that the bed is set too low.

If the BLTouch cannot trigger because the nozzle is already touching the bed, you have to change something on the bracket.

Did you follow the instructions?

wes1993, 2021-04-22:
Thanks a lot again for your reply @knutwurst 
Sorry bu i don't understand very well :-(
I have used the menu manual leveling and the manual leveling trigger the sensor, so i have manually leveled the bed based on Z=0 (Clicking on point A B C ecc).
So when i do tthis part here:
Slowly lower the nozzle with G1 Z-1 or G1 Z-0.1 or G1 Z-0.02 until the paper is difficult to move.
The paper is difficult to move because when Z=0 (endstop reached) because i have manually levelede the bed.

For BLTouch everything it's ok, just when i lower the nozzle i arrive at the endstop so z=0

In conclusion is possible that when i do the manual level i need to not put the paper? but in this way i have the problme that risk the bad scratch

Best Regards
Stefano

closed 2021-05-29 by github-actions[bot]