knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #130: MEGA_X_DGUS_TMC_BLT_10_v1.2.0-beta.1.hex bltouch won't stop

xinyuxh8247, 2021-04-19:
My mega X display is version 02, and the DGUS version is flashed in. There should be no problem with the Bltouch cable connection. The BLtouch feedback looks normal when I turn it on. When I turn on the autolevel to measure the first point, BLtouch touches the bed, but He couldn't stop, even if the Z axis had touched the limit switch, he still didn't stop until an error was reported. You can look at the video for details. 
[https://youtu.be/ZGjyM8l-02A
](url)
A friend of mine, his mega x display is an old version, and BLtouch can't stop after flashing in 1.2.0-beta.1, he finally chose 1.19, and everything is working normally.

I really don't know where the problem is, I really hope to get some help here.

knutwurst, 2021-04-21:
Hi! That's weird, but might be a firmware bug, if the problem does not happen in 1.1.9.

Also the Alpha 3 is not available for download anymore. Can you please try the 1.1.9 Final and the 1.2.0 Beta 1 afterwards?

malckris, 2021-04-21:
I see several mechanical problems in video.. 
- edge of nozzle is making too much contact with bed during homing process...> lower entire bed slightly
- BLTouch needs to be mounted lower, reprint BLTouch mount and make it longer...> nozzle is hitting bed before BLTouch activates
- adjust bed height or adjust limit switches to activate just after nozzle makes contact with bed...

knutwurst, 2021-04-22:
@xinyuxh8247 can you please verify that 1.2.0-beta.2 (or the provided hotfix) works for you? 

If you are not using an genuide Antlabs BLTouch, the issue should be fixed.

xinyuxh8247, 2021-04-22:
> Hi! That's weird, but might be a firmware bug, if the problem does not happen in 1.1.9.
> 
> Also the Alpha 3 is not available for download anymore for a good reason :)
> 
> Can you please test this hotfix version?
> 
> [MEGA_X_DGUS_BLT_10_v1.2.0-b1hotfix2.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6350164/MEGA_X_DGUS_BLT_10_v1.2.0-b1hotfix2.hex.zip)
> 
> EDIT: swapped hotfix1 to hotfix2 because ... yes... I needed to :D
> EDIT2: Do you use an genuine BLTouch Sensor or a clone?

I am using the cloned version of bltouch，A friend of mine also used the cloned version of bltouch. He reported that he had no problem with the firmware version 1.19. Using 2.0, the bltouch could not stop like me.

Thank you for your help anyway, I am going to test beta2 to see if there is any improvement.

xinyuxh8247, 2021-04-22:
> @xinyuxh8247 can you please verify that 1.2.0-beta.2 (or the provided hotfix) works for you?

I will arrange the test as soon as possible, and then give you feedback on the results.

xinyuxh8247, 2021-04-22:
> 我BLTouch的视频..通知，打印组件不回家过程中，使与床接触，也喷嘴时BLTouch被触发上述床..

Thank you for your feedback. In fact, after upgrading the 2.0 firmware, the four leveling screws of my entire bed have been pressed to the fullest. If the nozzle will directly touch the bed, I will return to the original firmware and reset to the zero position. The actual nozzle is still some distance from the bed, and the 4 leveling screws do not need to be pressed to the tightest position.

> @xinyuxh8247 can you please verify that 1.2.0-beta.2 (or the provided hotfix) works for you?
> 
> If you are not using an genuine Antlabs BLTouch, the issue should be fixed.

Thank you for your feedback. In fact, after upgrading the 2.0 firmware, the four leveling screws of my entire bed have been pressed to the fullest. If the nozzle will directly touch the bed, I will return to the original firmware and reset to the zero position. The actual nozzle is still some distance from the bed, and the 4 leveling screws do not need to be pressed to the tightest position.

And you mentioned the height of the bltouch problem. It seems that my installation should be no problem. I have already installed 3 1mm spacers. The current ejection state must be higher than the nozzle, and the retracted state is lower than the nozzle.

chrispa1606, 2021-05-23:
Hi, I am getting the same issue on my Anycubic Mega S. Have tried the original hex and also the bugfix you posted. everything seems to work fine except the ABL drives the head into the bed. I had another Marlin 2 firmware where it worked fine https://www.thingiverse.com/thing:4358979 but with that one I had temperature control issues on the hot end. My printer is the later model with the the 8 bit V0.0.2 running TMC2208's with the plugs reversed.

xinyuxh8247, 2021-05-25:
I tested the latest beta3 version, the fault persists, 3Dtouch still can’t stop, and then I have a latest discovery, using Gcode to control the 3Dtouch function through octoprint is OK, and I can’t stop the menu key limited to the display, my Z axis The stopper was damaged again. Fortunately, I bought a spare part and replaced it immediately.

chrispa1606, 2021-06-06:
Think I may have found the issue on this, went in and compared the working configuration.h file and recompiled from the base code. Not 100% but think the only value I really changed that worked was #define Z_MAX_ENDSTOP_INVERTING true as it was set to false in the original hex. I have it installed and it seems to be working great. If it isn't this and I changed some other values I can attached the working configurtion.h file just in case this isn't the fix.

.

knutwurst, 2021-06-29:
If Z_MAX_ENDSTOP_INVERTING works for you, then you did not wire up the BLTouch correctly. Also the ZMAX isn't used at all. 

It must be an electrical issue.

closed 2021-08-05 by github-actions[bot]