knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #131: BL-Touch Problem

ar84, 2021-04-19:
Hallo 

Ich habe das Problem das beim Auto-Bed-Leveling, egal ob über Display am Drucker oder per Repetier, der Sensor an die erste Position fährt 2x ein aus fährt und dann stehen bleibt.

Was kann ich tun?

knutwurst, 2021-04-21:
Hallo,

so gern ich helfe und so leid es mir tut, aber dies ist ein Firmware-Bugtracker für Fehler in der Firmware. Das von dir geschilderte Problem ist zu 99% hausgemacht durch einen nicht korrekt eingestellten Z-Offset oder einem defekten Sensor. 

Diese Frage wurde aber schon mehrfach sowohl in der BLTouch Anleitung als auch in den FAQ behandelt. https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki

ar84, 2021-04-21:
Hallo.
Sorry hatte vergessen den Post zu schließen. Fehler gefunden.
Gruss

closed 2021-04-21 by ar84