knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #133: [BUG] BLTouch 3.1 issues

AndrewLBC, 2021-04-23:
Last night i installed a brand new BLTouch 3.1 from an authorized legit reseller as well as updated my firmware to  MEGA_S_TMC_BLT_11_v1.2.0-beta.2.hex
I followed this guide on the installation. https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(english)

When i power the i3 Mega S up, the BLTouch has a red light come on and it the probe extends back and forth and other than that i can't get it to do anything aside from the light turns off when i manually pull the probe out. I've tried the M280 command to deploy the pin but it just sits there with the red light on solid. Apparently it also has a blue light but i have yet to see that at all. 

I'm fairly certain my Trigorilla board is the 1.1 type since it says 0.0.2 under the logo, the large connectors are green, the S1 row of servo pins are populated, which is where i hooked up the BLTouch, along with the two soldered on pins on the little PCB up in the hotend housing. 

Any help would be greatly appreciated. Thanks!

knutwurst, 2021-04-30:
Hi,
0.0.2 doesn't mean anything. As described in the guide, please use the _10 first, and only if this doesn't work, go for the _11.

If the problem persists, check your soldering ;) 
That's the most common problem. Also, if you really tried everything, you can try the Beta 1, since there I used an (unstable) software configuration especially for the BLT 3.1. But it breaks other sensors, so I reverted the code.

AndrewLBC, 2021-05-02:
Ah. I had read my 0.0.2 board was version 1.1 and required the firmware i mentioned above. Ill load _10  and see if it is a solution. 

As for the soldering, i'm confident that is fine. I double checked everything with a multimeter. Hopefully that will fix it since i'm just finishing up building an enclosure for the printer and manual bed leveling inside the box will suck.

knutwurst, 2021-05-14:
Hi @AndrewLBC, is there still an issue?

AndrewLBC, 2021-05-17:
I'll know by tomorrow. Been out of town and have my printer in pieces due to upgrading the entire hot end assembly to that X-Carriage MK4 and a custom built enclosure for everything. I'm assembling the whole thing right now. 

I think the problem is my 0.0.2 board which has the servo pins soldered on S1 might actually be a 1.0, and i was using firmware for the 1.1. When i get it assembled, i'll be putting firmware for the 1.0 board and see how it goes.

AndrewLBC, 2021-05-18:
VICTORY!!! I Reassembled the printer, installed MEGA_S_TMC_10_v1.2.0-beta.3.hex and followed your setup guide and my BLTouch now works perfectly. Well... there was one slight issue where i forgot i had to calibrate it and ran auto leveling and the 5th point it crashed into the bed. lol. oh well. 

Almost done with the enclosure. Venting the electronics from below so they don't get too warm. Still gotta figure out an air filter for the print area. https://i.ibb.co/vCNzjd0/Printer-Enclosure-In-Progress.png

Gonna test print something then get some sleep. Thanks again.

knutwurst, 2021-05-18:
That looks great! I'm glad that everything works now! :)

closed 2021-05-18 by knutwurst