knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #134: [Request] Heating Failure

Jack77777777, 2021-04-30:
Hi Knut,

I updated from an older Marlin to your **MEGA_S_TMC_BLT_11_v1.1.9** version. But getting heating failed messages and print stops. Downgrade to the older version fixes the issue.

I assume changing some values in Configuration_adv.h could fix the issue. See values from working older version and from your version: 

**Working older Marlin**
```
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 60        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 20     // Degrees Celsius
  #define WATCH_TEMP_PERIOD 60                // Seconds
  #define WATCH_TEMP_INCREASE 3               // Degrees Celsius
  
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 20    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 5 // Degrees Celsius
  #define WATCH_BED_TEMP_PERIOD 60                // Seconds
  #define WATCH_BED_TEMP_INCREASE 2               // Degrees Celsius
```
  
**Your version**
```
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 10        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 15     // Degrees Celsius
  #define WATCH_TEMP_PERIOD 30                // Seconds
  #define WATCH_TEMP_INCREASE 3               // Degrees Celsius
  
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD        20 // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS     2 // Degrees Celsius
  #define WATCH_BED_TEMP_PERIOD                60 // Seconds
  #define WATCH_BED_TEMP_INCREASE               2 // Degrees Celsius
```

I would compile by myself but i don´t know what you changed for the different version. Could you provide me the Source Code for **MEGA_S_TMC_BLT_11_v1.1.9** or compile it with the above values ?

knutwurst, 2021-04-30:
Hi :)
My name is not Knut, but Oliver;)

Anyway... your configuration only considers a time window of one minute and expects an increase of 3°C with fluctuations of up to 20 ° C. In addition, there is a long wait until errors are triggered. This means that your configuration doesn't really care if the temperature stays stable.

Of course, if that's what you want, you can compile it yourself.

The source code for 1.1.9 is included with the release as a .zip. Alternatively, you can use the current version from the master. It is a unified version that fits all printers.

Please see the wiki for instructions on how to use PlatformIO, because I use a lot of features from PlatformIO so that you can no longer compile it with the Arduino IDE.

Also please read the FAQ, since all your Questions are already answered there. 

Best
Olli

Jack77777777, 2021-04-30:
Sorry **Oliver**,

Thank you pointing me to the FAQ.
Decreased FAN to 70%, did Auto PID Tuning for Hotend and Bed using the Display (Formerly i used terminal to do that).
Seems to work now.

Thank you

Jack77777777, 2021-05-07:
Above fixed the issue. This can be closed. 

Thanks a lot!

closed 2021-05-14 by knutwurst