knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #139: [FR] (Please update part links on )

bassamanator, 2021-05-11:
### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2021-05-17:
Hi @bassamanator ,
since I got mostly negative feedback on affiliate links, I removed them completely ... I just forgot to do so on the english translated page. Sorry.

Yes, the Link you provided ist totally fine. Everything should work as expected. :)

closed 2021-05-17 by knutwurst

bassamanator, 2021-05-25:
I'd like to integrate you into my aliexpress and forget about it. How do I do that?