knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #140: BL-Touch - Probe is not fully triggering

Blackhawk50000, 2021-05-12:
I am pritty Shure, that this is not a bug, just PEPCAK.
I did the tutorial for setting up and installing everything step by step.

i cablemanaged everything and solderd what i should solder. and almost everything works fin. i can control the BL-Touch via G-Code (i guess)

so i after installing the BL-Touch i followed setting it up with the G-Codes. i was a little bit confused on step 17 but i finally figured it out and the result was 2,34 mm (2,40-0,04)

but after starting the BL Touch Bed Leveling, the hotend goes down and down. the BL-Touch i blinking BUT noch triggering. the hotend goes down even further. so far, that the nozzle pushing down the heatbed. 

what could. what have i done wrong?

cryoNZ, 2021-05-15:
swap your black and white wires around. They are in reverse so its not detecting the trigger.

knutwurst, 2021-05-15:
As @cryoNZ said, it must be some hardware/wiring issue. This has nothing to do with the firmware itself. Please see: #69

SteFiD93, 2021-05-18:
stupid question .. have you tried to change the version of the trigorilla in the firmware .. from 11 to 10?

Blackhawk50000, 2021-05-20:
> swap your black and white wires around. They are in reverse so its not detecting the trigger.

this was the thing.
i thought that i read the guild carefully enought but obviosly i got confused anyway in this thing... now it works

closed 2021-05-20 by Blackhawk50000