knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #141: Anycubic Chiron-Bauteilkühler von Fan0 auf Fan1 legen-Place component cooler from Fan0 to Fan1

ronaldi40, 2021-05-14:
Hallo, ich besitze einen Anycubic Chiron und brauche dringend Eure Hilfe. Mein Fan0 Augang ist defekt. Ich habe den Bauteilkühler an Fan1 angeschlossen. Unter der alten Firmware v1.1.9 war das kein Problem mit Arduino (pins_RAMPS.h) den Ausgang auf Fan1 zu legen. Mit der neuen Marlin 2 von Dir und PlatformIO komme ich überhaupt nicht klar. Schon alles mögliche ausprobiert. Deine fertig kompilierte Firmware HEX funktioniert wunderbar, aber meinen Fan1 Lüfter kann ich einfach nicht steuern. Entweder läuft er volle Pulle oder gar nicht. 
Nun würde ich sehr gerne die Einstellungen wissen, um die Marlin Firmware so zu ändern, das FAN0 stillgelegt (da defekt) und FAN1 statt dessen aktiviert wird. Bitte helf mir, ich bin am verzweifeln.
Hello, I own an Anycubic Chiron and I urgently need your help. My Fan0 output is defective. I connected the component cooler to Fan1. With the old firmware v1.1.9 it was no problem to connect the output to Fan1 with Arduino (pins_RAMPS.h). I can't handle the new Marlin 2 from you and PlatformIO at all. Tried everything possible. Your fully compiled firmware HEX works wonderfully, but I just can't control my Fan1 fan. Either it runs full or not at all.
Now I would like to know the settings to change the Marlin firmware so that FAN0 is shut down (because it is defective) and FAN1 is activated instead. Please help me, I'm desperate.

knutwurst, 2021-05-15:
Hallo @ronaldi40,
schau mal bitte im [Wiki](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki) nach. Dort findest du eine kleine Anleitung + die offizielle verlinkte Anleitung zum Arbeiten mit PlatformIO.

Mit der Arduino IDE kannst du leider nicht mehr arbeiten :/ 

Ich möchte dir daher ans herz legen, dich einmal in PlatformIO einzulesen, da du dann viel flexibler arbeiten kannst. Nichtsdestotrotz habe ich dir mal eine Version gebaut, bei der ABL deaktiviert ist (damit du mit MBL leveln kannst) und FAN0 (D9) gegen FAN1 (D7) getauscht wurde. Dein Ticket #142 habe ich in dem Zuge mal geschlossen.

Bitteschön: 
[CHIRON_120_b3_fan01.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6488086/CHIRON_120_b3_fan01.zip)

closed 2021-05-15 by knutwurst

ronaldi40, 2021-05-15:
Hallo Knutwurst, erst einmal vielen Dank für Deine Antwort. Ich bin der Meinung, das Deine Version die Beste ist.
Nun meine Bitte, ich würde gerne die integrierte Anycubic Chiron - Auto-Levelling Funktion (ABL) weiter benutzen. Leider funktioniert diese mit keiner Version von Dir, weder die Beta, auch nicht die Master. Ich habe sie alle ausprobiert. Bleibt bei "Probing..." stehen und das war es.
Kannst Du mir noch einmal helfen?
Hello Knutwurst, first of all thank you very much for your answer. I think your version is the best.
Now my request, I would like to continue using the integrated Anycubic Chiron - Auto-Leveling Function (ABL). Unfortunately this doesn't work with any version of yours, neither the beta nor the master. I've tried them all. Stop at "Probing ..." and that's it.
Can you help me again

knutwurst, 2021-05-15:
Hi, nein, sorry.. wie schon in #144 beschrieben ist das aktuell ein Bug und ich arbeite noch an dem ABL feature. Das ist relativ schwierig umzusetzen, weil ich noch keinen eigenen Chiron besitze. Bitte hab noch etwas Geduld :)