knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #143: [BUG] Current 5 mesh-Points with bltouch runs out of bed with mk4 x-carriage

hawkeyexp, 2021-05-14:
While using MK4-Mod with bltouch and 5 mesh points are set leveling is done from left to right in x and last mesh point is out of bed.
Reducing to 4 mesh points work's well and lveling starts from right to left (x) and all works fine.

Possible a hint in the config would be helpfull or a check in config which limits mesh points.

knutwurst, 2021-05-14:
This is not an issue in the firmware or an issue with the number of measuring points, but your probe offset for the X-Carriage MK4 is not set. Please use M851 to set the offset correctly. Then the sensor is no longer outside the bed.

closed 2021-05-14 by knutwurst

hawkeyexp, 2021-05-14:
i have done it but i will ceck it again (x+29 y-15 commented in inside config)