knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #144: Anycubic Chiron - The auto-leveling function (ABL) does not work with any version. I've tried them all. Stop at "probing" and that's it.

ronaldi40, 2021-05-15:
Anycubic Chiron - The auto-leveling function (ABL) does not work with any version. I've tried them all. Stop at "probing" and that's it.
Anycubic Chiron - Die Auto-Levelling Funktion (ABL) funktioniert mit keiner Version. Ich habe sie alle ausprobiert. Bleibt bei "Probing" stehen und das war es.

knutwurst, 2021-05-15:
That's a known issue and since the 1.2.0 is in beta state, bugs can occur. Thanks anyway!

ronaldi40, 2021-05-15:
Hallo, schade, gibt es eine Version, wo das ABL funktioniert. Ich wollte Deine Version wegen des Mehrwertes benutzen, sprich Filamentsensor aus etc. Das bringt aber leider nichts für mich, ohne ABL.
Ach so, ich habe mich jetzt etwas in VSCode, PlatformIO eingelesen und das Problem mit meine Fan1 so gelöst:
1.) Folgende Datei in Marlin 2  in VSCode-PlatformIO bearbeiten:

      pins_TRIGORILLA_CHIRON.h in Verzeichnis Marlin\scr\pins\ramps

      #define TG_FAN0_PIN                      7  // Anycubic Chiron: Usually the part cooling fan
      #define TG_FAN1_PIN                      9  // Anycubic Chiron: Unused
      #define TG_FAN2_PIN                     44  // Anycubic Chiron: Hotend fan
      #define CONTROLLER_FAN_PIN               9
dank Deiner pins_TRIGORILLA_CHIRON.h
Jetzt fehlt nur noch die ABL.
Gruß
Ronald

ronaldi40, 2021-05-15:
Hi, in dieser Version v1.31
https://www.thingiverse.com/thing:3593131
funktioniert das ABL. Vielleicht kannst Du ja damit etwas anfangen.