knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #146: [BUG] (Filament Load/Unload didn't work)

Schnuecks, 2021-05-18:
### Bug Description

Hi, i installed the 1.2.0-beta.3 on my Mega Pro

Filament load via menu is not possible.

If I press Filament In or Filament Out it shows that it must heat, but nothing happens after that.

sanangeles84, 2021-05-19:
I also have the same issue, same firmware same machine, I have to raise the temperature manually on my own (typically to 220°C), then filament IN and OUT will work, original firmware starts the warm up automatically after the warning/error

HaGueRe, 2021-05-19:
Same issue on Mega-S. Also no raise on z-axis when pressing Filament In or -Out.

9V1LH, 2021-05-19:
Same here with MEGA_S_DGUS_TMC_BLT_10_v1.2.0-beta.3. It was still working with MEGA_S_DGUS_TMC_BLT_10_v1.1.9 .

knutwurst, 2021-05-20:
So the problem only occurs on 1.2.0 beta 3? What about beta 2?

Schnuecks, 2021-05-20:
On beta 2 it works normal.

9V1LH, 2021-05-20:
> 
> 
> So the problem only occurs on 1.2.0 beta 3? What about beta 2?

I can also confirm, that it still worked in MEGA_S_DGUS_TMC_BLT_10_v1.2.0-beta.1.hex and MEGA_S_DGUS_TMC_BLT_10_v1.2.0-beta.2.hex

HaGueRe, 2021-05-20:
Also for my Mega-S. Beta 2 and beta 1 of 1.2.0 are working fine

knutwurst, 2021-05-20:
That's weird... but I'll have a look ;)

knutwurst, 2021-05-31:
I don't own that exact type of printer.. could you test this version, if the filament load/unload function works as expected?
[MEGA_S_DGUS_TMC_BLT_10_v1.2.0-beta.4-pre.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6569758/MEGA_S_DGUS_TMC_BLT_10_v1.2.0-beta.4-pre.hex.zip)

Schnuecks, 2021-05-31:
Hi, on my Mega Pro it starts heating now on load and remove.

knutwurst, 2021-05-31:
Great, thank you!
What about you @HaGueRe , @9V1LH and @sanangeles84 ?

sanangeles84, 2021-05-31:
Hi @knutwurst , I'll try out this evening when I finish working.  Just as confirmation, I don't lose the Mesh leveling saved in the EEPROM when I flash a new firmware, right?

knutwurst, 2021-05-31:
Nope, every firmware since KW-1.1.6 uses the same eeprom layout, so it's compatible

9V1LH, 2021-05-31:
> I don't own that exact type of printer.. could you test this version, if the filament load/unload function works as expected? [...]

Works fine now. Great, thanks!

HaGueRe, 2021-05-31:
Just printing, i will try this evening. Can i use the pre.hex for my i3 Mega S without BLT just for a try?

HaGueRe, 2021-05-31:
The printjob is finnished and i gave it a try. "Filament In" and "Filament Out" are working fine :). I had to lift up the z-axis before the flash, cause my printer has stock motor drivers, and, as expected,  my print head goes down instead up when hitting the button.

Many thanks for your efforts!

knutwurst, 2021-05-31:
Great - This will be added in further releases. The fix has been applied to the current master branch. Thank you very much for submitting the bug!

closed 2021-05-31 by knutwurst

sanangeles84, 2021-05-31:
Sorry for being late, I can confirm as well, it works the warm up also for me and both for filament in and remove
@HaGueRe thanks for your feedback, you avoided me to hit the bed like I did yesterday, which forced me to recalibrate everything from scratch  :)

Schnuecks, 2021-06-01:
I have to correct, the heat starts that's ok but Filament In (unload) and Filament Remove (load) tried it multiple time with full reset.

sanangeles84, 2021-06-01:
I honestly haven't tried to move the filament in and out, just if it was warming up, so warms up ok, moving filament I don't know.
For z-axis direction, also mine was upside down but it was just because of my Mega Pro version (and I suppose also yours) that should have the TMC motors already with inverted plugs, so that shouldn't be an issue, it's just that he sent us the "non-generic" version (the one without TMC in the name) that inverts the z-axis

Schnuecks, 2021-06-01:
@sanangeles84 thank for this info.

sanangeles84, 2021-06-01:
@Schnuecks So, since I didn't test till that point, does it move the filament when you press again those options after it reaches the 220°C?

Schnuecks, 2021-06-01:
No, I could press multiple times on In it unloads and vice versa.

sanangeles84, 2021-06-01:
I'll try again on mine during lunch break, also moving filament this time, let's see what happens on mine

sanangeles84, 2021-06-01:
Just tried, after warming up to 220-230 I'm able to move the filament, it's just that also (like for z-axis) in this case movements are inverted, "filament in" removes it and "filament remove" lets the filament go in, 
@knutwurst is this filament movement inversion (in->removes and remove->let in) again related to TMC_ firmware or a bug?

HaGueRe, 2021-06-01:
As i said above all works fine. on my i3 mega s with stock motor drivers. **All** stepper motors are running inverse as exspected with the tested pre.hex. You  have still have to wait until  the hotend reaches a temperature > 180°C to move the filament

sanangeles84, 2021-06-01:
So also for me everything works as expected, my Mega Pro motors have been built with inverted plugs so it works like a Mega S

knutwurst, 2021-06-01:
Hi ;)
yes.. the _TMC flag indicates that all 5 stepper drivers must be changed to Trinamics. If not, all axes will be in reverse. 

The normal Mega Pro only has 2 TMC drivers (X and Y) and therefore no _TMC is needed, but if the plugs are inverted and it behaves like a normal Mega S (with, or without TMC) the _S firmware should be suitable.

I built this particular one only because I've seen that @9V1LH used the MEGA_S_DGUS_TMC_BLT_10_v1.2.0 Version and built the hotfix with these settings.

As I mentioned... feel free to use the Beta 2 or the 1.1.9 instead which should work as expected. The bug in the Beta 3 is reverted/fixed.

sanangeles84, 2021-06-01:
I'm fine with the Beta 3 for now, I just need to change manually the temperature before trying to move the filament, it works as workaround.

HaGueRe, 2021-06-01:
Maybe it's my fault, but is there anyone else who lost his eeprom datas after flashing the self compiled new code?
No problem for me, cause i made a backup of my level-data, put it in a gcode and saved it to eeprom after "printing" this file.
The first print is running and for now all is fine :)

But, if your printer stops printing after homing, mention this.

sanangeles84, 2021-06-01:
Unfortunately I also lost my level-data, and I didn't have any backup. Do you @HaGueRe have by chance any guide/website/video that I can follow to create a backup and gcode as well? I don't want to risk again to lose level-data, just tried to level again and I didn't do it properly, it's so timewasting to do it manually each time and I'm not planning to move to the automatic levelling yet.

HaGueRe, 2021-06-01:
Shure :)
I took this instruction from David Ramiro, a other  firmware for Anycubic Printers (Even good, but not so actual as the one by knutwurst):

_Updating
Back up & restore your settings
Some updates require the storage to be cleared (M502), if mentioned in the update log. In those cases, before updating, send M503 and make a backup of all the lines starting with:

M92
G29
M301
M304
After flashing the new version, issue a M502 and M500. After that, enter every line you saved before and finish by saving with M500._

Put all these lines into a textfile, call it "backup.gcode" and "print" it. Send M500 to printer (you can also put M500 as last line into the gcode).

For my(!) printer it looks like this:

_M502
M500
M92 X80.00 Y80.00 Z400.00 E384.00
G29 S3 I0 J0 Z-0.04000
G29 S3 I1 J0 Z0.10000
G29 S3 I2 J0 Z0.04000
G29 S3 I3 J0 Z0.02000
G29 S3 I4 J0 Z-0.04000
G29 S3 I0 J1 Z-0.08000
G29 S3 I1 J1 Z0.04000
G29 S3 I2 J1 Z0.02000
G29 S3 I3 J1 Z0.02000
G29 S3 I4 J1 Z-0.04000
G29 S3 I0 J2 Z0.08000
G29 S3 I1 J2 Z0.06000
G29 S3 I2 J2 Z0.08000
G29 S3 I3 J2 Z0.02000
G29 S3 I4 J2 Z-0.04000
G29 S3 I0 J3 Z0.02000
G29 S3 I1 J3 Z0.10000
G29 S3 I2 J3 Z0.12000
G29 S3 I3 J3 Z0.10000
G29 S3 I4 J3 Z0.02000
G29 S3 I0 J4 Z0.26000
G29 S3 I1 J4 Z0.16000
G29 S3 I2 J4 Z0.18000
G29 S3 I3 J4 Z0.14000
G29 S3 I4 J4 Z0.06000
G29 S4 Z0.00000
M301 P15.94 I1.17 D54.19
M304 P251.78 I49.57 D319.73
M500_

Greetings :)