knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #148: [Fehler] (kurze Beschreibung)

Pkahl, 2021-05-22:
Ich habe eine Anycubic-i3-Mega-s Ich Habe ein BLTouch eingebaut Aber nicht verlötet Ich Habe die Auf den Board gesteckt Weiß Auf SCL Ping 21 und den Schwarz auf T3 und die 3 Kabel ist am Servos angeschlossen Und der BLTouch Funtioniert der Blinkt und Fährt auch aus Aber beim Auto Leveling Home Der Und Fährt auch Hoch und geh zu den ersten Punt Aber der fährt nicht Nach unten. Und Bei der Konsole Ist einen Fehler 
Ich Habe den Pin Von Weiß Von 2 Auf 21 Geändert Kann einer Helfen ???

Recv:12:48:14.350: X:0.00 Y:0.00 Z:0.00 E:0.00 Zählen X:0 Y:0 Z:0
Recv:12:48:27.610: Fehler:!! STOP wegen BLTouch-Fehler aufgerufen - Neustart mit M999
Recv:12:48:27.611: Error:Probing Failed
Recv:12:48:27.613: X:12.00 Y:35.00 Z:21.18 E:0.00 Anzahl X:960 Y:2800 Z:8472
Recv:12:48:27.649: echo:Einstellungen gespeichert (712 Bytes; crc 50839)

https://www.youtube.com/watch?v=Na1ttOOhRcc

Jack77777777, 2021-05-26:
Welche Firmware hast Du installiert?

Pkahl, 2021-05-28:
Ich habe den Mega_S_BLT_10 Mit PlatformlO Installiert Weil ich den Pin ändern musste von 2 auf 21 Von Weiß

Jack77777777, 2021-05-28:
Sicher, dass Du Dein Board die Version 1.1 hat?

Pkahl, 2021-05-29:
Das Weis ich nicht Genau Aber ich Habe beide Mega_S_BLT_10 und Mega_S_BLT_11 Ausprobiert und Beide Zeigen den Gleichen Fehler An

Jack77777777, 2021-05-30:
Dann bin ich leider überfragt.

knutwurst, 2021-05-30:
Wenn der Sensor ausfährt (mit gcode), dann stimmt das so mit der BLT_10 Version. Hier unterscheidet sich nur der Servo-Port, der die Spitze ausfahren lässt. 

Der Fehler kommt vermutlich vom nicht (oder falsch) eingestellten Offset.

Sind die Defaults geladen? Ist der Offset richtig eingestellt? Wurde die Anleitung befolgt? An welchem Punkt der Anleitung scheitert es?

Welche Version vom Sourcecode verwendest du?

Pkahl, 2021-05-30:
Ich habe die Anleitung befolgt bis zum Schluss und alles hat auch gut funktioniert aber ab den punkt wo ich den knopf drücke zum leveln ab da bekomme ich den Fehler immer aber ich Habe den  Configuration.h Datei von Dieser Schnee rauskopiert von https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/pull/61/commits/606c649f28d48bf0dcc860e630c3877e2ad5da77 und ab ab jetzt funktioniert alles

Pkahl, 2021-05-30:
Eine Kleine Frage kann man den gebauten Datei von PlaftormlO Auch zu einen Datei erstellen Das ich nicht mit USB mit den Drucker Verbunden sein muss Wegen der Drucker hängt am Server

ALEX9886, 2021-06-03:
Hi, 
ich habe genau denselben Fehler. Mein BL-Touch v3.1 fährt an die erste Position und löst mehrmals in der Luft aus. Er fährt nicht nach unten und das Programm läuft nicht weiter. Anschließend kommt das "probing failed".
@Pkahl konntest du das Problem lösen?
@knutwurst ich habe auch schon die beta v2 ausprobiert. 

Woran liegt das? Angeschlossen habe ich alles nach der Anleitung. Offset ist auch sicher richtig.

Pkahl, 2021-06-04:
Ich habe dem Problem gelöst ich habe aber einfach gemacht ich habe den Configuration.h vom https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/606c649f28d48bf0dcc860e630c3877e2ad5da77 herauskopiert und bei mir eingefügt und bei dem Pinbelegung habe ich nur geändert weil ich das dem Mainboard gesteckt habe nicht gelötet habe da hat alles gut funktioniert

closed 2021-06-29 by knutwurst