knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #149: [BUG] Unknown board - no info on wiki

JernejL, 2021-05-24:
### Bug Description

Trouble finding wiki info - bltouch wiki page says i have to use correct version ( 10, 11 ) but there is no info on how you actually find out the version on the wiki, i have a i3 mega X bought in january this year. 

### Steps to Reproduce

1. https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(english)
2. There is a chapter about "It is important to note which board you have (TRIGORILLA_14 v1.0 or v1.1), since the pin assignment has changed here."
3. There is no info on how to find out which board you have. 

**Expected behavior:** Page with pictures or some kind of instructions

**Actual behavior:** No info :( 

#### Additional Information

* See https://i.imgur.com/3ZXK70O.png - the only marking about version i could find is this, it says "trigorilla board v 0.0.2" so it seems not to match anything you have on wiki.

Jack77777777, 2021-05-26:
You can find it out if you follow this:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(english)#install-the-firmware

JernejL, 2021-05-26:
This is bizzare, can i find out version in another way? using picture comparison or gcode command?

knutwurst, 2021-05-26:
Unfortunately, there is no official way to find out which motherboard it really is, as Anycubic has not documented this. Sometimes the label says 1.0 or 0.0.1 or 0.0.2. Unfortunately, this does not provide any information about the pin assignment of the servo port.

The easiest way for me (as also described in the instructions) to try it out. First with the _10 and then with the _11.

Datafreak, 2021-05-29:
I had the same issue with sa simmilar 0.0.2 board.
The FIrmware with "_10" was correct for mine.

JernejL, 2021-06-02:
Main issue i have is that instructions say that that pin assignment has changed between those trigorilla board versions, so if i can't find out which board i have it will fry bl-touch if not connected properly? there is no clear picture in tutorial on how the servo port looks like on the board and how wires get connected to it. 
 
Should i just orient cabling based on markings on board (if i find any) and then try one or the other firmware versions?

knutwurst, 2021-06-02:
No, nothing happens. The assignment for the servo pin that has already been soldered has only changed. In the worst case, the BLTouch just won't work. So when you do test 2, the tip has to be extended. If that doesn't happen, just take the other firmware.

JernejL, 2021-06-10:
I've started actual installation now and will follow most of the manual. 
 
The STL to mount bltouch linked to on thingiverse printed well, however when i mounted the bltouch to it the probe point did not protrude further than the nozzle, is this maybe some new issue? 
 
Bltouch is not original but clone. i've added M3 nuts to get it to protrude further, do i need to adjust some configuration for this?

Datafreak, 2021-06-10:
The linked [Mount](https://www.thingiverse.com/thing:2824005) for the STL is for older BL-Touch Sensors.

You could try: [Anycubic I3 Mega BLTouch V3 Mount](https://www.thingiverse.com/thing:3660135) for newer BL-Touch Sensors like v3.1
Or one of the other Remixes wich are linked on Thingiverse.

JernejL, 2021-06-10:
Is there anything to adjust if i just put in 2 washers? seems like tutorial will need update too :)

JernejL, 2021-06-14:
Thanks @Datafreak , I've printed the other mount that you suggested and that worked perfectly. 
 
In the end i used _10 firmware and also had to use DGUS version that is not in stable release, so i'm now running MEGA_X_DGUS_BLT_10_v1.2.0-beta.4 and this worked well with my version. 
 
Here is my wiring, i hope this is clear as you can follow wire colors from bltouch to the head and also to the board itself. 
 
[image]
[image]
[image]
[image]

closed 2021-07-20 by github-actions[bot]