knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #150: [BUG] Offset on Y Axis after 38mm high

dobbidev, 2021-05-27:
### Bug Description
I am not sure how to describe the behavior correctly:
After 38 mm on Z, the print get an offset of 10-12 mm on the y axis.
I tested it two times, with other slice settings. The Bug is the same. 
Had anybody something similar so far? 

### Steps to Reproduce
Here is my used gcode file:
[AI3M_LiebherrFridgeMoutingBracket.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6557043/AI3M_LiebherrFridgeMoutingBracket.zip)

**Expected behavior:**
Like this one: https://www.thingiverse.com/thing:2753818

**Actual behavior:**
[image]

#### Additional Information
I use an Anycubic i3 Mega M
With the latest Release Version 1.1.9

dobbidev, 2021-05-29:
Hello,

I did some test prints  the last day and no model had the same bug like in the image on the top.
I printed it now on the side:
[image]
[image]
and see -> the offset is not longer existing...

I have no idea how this can happen, but it looks like something has to went completly wrong at the slicing step. 
I hope that error does never come again up :)

Have a nice day!

closed 2021-05-29 by dobbidev

knutwurst, 2021-05-29:
Hi @dobbidev ,
This is a common problem when the motor drivers overheat or fail to deliver the required power to the motors at high temperatures.

I generally recommend operating the TMC2208 with at least 1 volt Vref, otherwise layer shifts will occur. With the Y-axis you can go up to 1.3V, as more power is needed here.

It looks like you only had the problem on the Y axis and I just assume that you installed TMC2208 drivers.

These fan ducts with axial fans are also extremely counterproductive. Axial fans cannot build up air pressure (only radial fans can do that) and so the air suppresses itself. That is why a fan cools much worse than the original fan. As a result, TMC drivers can also overheat and layer shifts occur.

The fact that the layer shift always occurs in the same layer is simply due to the fact that at this very moment the bed has to change direction at an unfavorable angle and time and this is where too much power is being demanded from the motors.

Solution:
- Higher reference voltage = more possible motor current
- Better cooling of the drivers

I'll update the Wiki ;)