knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #151: [BUG] ABL data wrong or not correct used

Datafreak, 2021-05-29:
### Bug Description

I have the feeling, that the ABL Data are collected in a wrong way or will not be used correctly. My printer will not produce a even first layer.
I installed a Antclabs BLTouch V 3.1 on my Anycubic i3 Mega S. The Firmware I had to compiled it myself, because I use PIN 21 (MEGA_S_BLT_10).
The bed is leveled with the the Angle Correction of the Octoprint Bed Visualizer Plugin.
When I print a first layer the left side is thinner then the right one. I have the feeling, that the feeling, that the ABL data are fliped or somingelse is going on. I could not find the reason.

### Steps to Reproduce

1. compile & install Firmware 1.1.9 or 1.2.0-beta.3
2. level the bed with Angle Correction of the Octoprint Bed Visualizer Plugin
3. do a BLTouch Leveling via the Special Menu
4. print the [Anycubic i3 Mega Bed Level](https://www.thingiverse.com/thing:3244876)

**Expected behavior:** first layer should be even

**Actual behavior:** left side is thinner

#### Additional Information

* [Image of my bed mesh](https://imgur.com/NnRETdF)
* [Image of the printed result](https://imgur.com/ZBETGMk)

MaGe-1, 2021-05-30:
Hi, 

i am also facing a similiar problem.
With ABL i am not able to print a good first layer. Using Mega-x without modifications despite the abl

It seems like the mega-x is compensating the values in a wrong manner. cant describe how. sometimes it is much to far away from the bed and printing into the air.

Kind regards
MaGe

dotkoval, 2021-06-03:
same here. spent several evenings trying to re-calibrate and doublecheck everything: z-offset, manual bed level and springs, x-axis alignment (z-sync), both z-endstops (z_min and z2_min). tried enabling extra probing and 7x7 mesh. and on printing bed level test always getting left side lower then right. like bed is tilted a bit. but mesh coords looks fine - comparable and close enough to results i've got with MBL previously.

MaGe-1, 2021-06-03:
Maybe the sequence of measurement is incorrect mapped to the mesh points?

dotkoval, 2021-06-04:
Mesh seems to be completely valid. No mesh points displacement or something like this. Height map also seems to be valid - was comparing values with manual absolute positioning and measurments got during G29/G30. Mesh is active and I see that correction applied during the prints (z-axis moves a bit). Ran out of ideas.

MaGe-1, 2021-06-04:
My z axis is also moving. But i habe the problem, that it is Not working aß expexted. I have problems with the first layer, which is Not connected in a food way to the bed. Because in some areas IT is too close or too far away from the bed. 
I think IT is a softwareproblem, But i dont know how to isolate the error.

dotkoval, 2021-06-04:
I spent almost all the day trying to figure out a pattern... I performed 7x7 MBL with perfect 1st layer printed and compared it to ABL measurements. Looking into visualized map I have guess, that it might be connected with ignoring probe to nozzle offset. I have BLTouch mounted on right side (X29.00 Y-15.00), what the offset in your case, @Datafreak? Maybe with some geometry and offset X close to 0 firmware issue just not so recognizable.
I found a similar Marlin issue discussed  (https://github.com/MarlinFirmware/Marlin/pull/20290), but it does not look like a case (I've check the code - seems fine, but I'm not fimilar with Marlin code and knutwurst changes, may miss something)

[image]

Datafreak, 2021-06-04:
Hi @dotkoval,

I use this mount, because I have a BLTouch V 3.1:https://www.thingiverse.com/thing:3660135

NOZZLE_TO_PROBE_OFFSET is: { -2, -25, 0 }

Currently I try to get my Ancubic running with Klipper to print with 250mm/s.
So I cant play with Marlin and the Offset. But when I have a working Klipper Configuration, I will switch back to Marlin and messure the offset myself and set it in the configuration.

Best regards,
Datafreak

dotkoval, 2021-06-05:
Thanks @Datafreak!
btw, it would be great to know, is similar BLTouch issues on your
Klipper setup happens. just to exclude hardware part and be sure, that
it is firmware issue. 

>

dotkoval, 2021-06-18:
i think it is mechanical issue and not a firmware bug.
it is possible, that X-axis rods not well aligned and the further from the nozzle sensor placed - it gives more error.
aligning these rods a bit tricky, but i guess that's the main reason.

knutwurst, 2021-06-21:
That's a mechanical issue and there is also a (software) fix for that: https://habitoti.de/wordpress/bltouch-frustration/

closed 2021-06-21 by knutwurst