knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #153: BLTouch (CN clone 3Dtouch)  an Anycubic I3 Mega Pro/Tastpin fährt nicht aus

faitesvosjeux, 2021-05-31:
Anycubic I3 Mega Pro, Mainboard Trigorilla V0.0.2, KW MEGA_P BLT 1.2 beta 3/MEGA BLT 11 1.2 Beta 3

3DTouch nach Anweisung mit Halter aus Thingiverse verbaut, Verkabelung zu D11 (auf Mainboard "S1 " beschriftet) gelegt, auf Polarität geachtet und 5V/GND entsprechend vertauscht. 3DTouch fährt nach Einschalten Pin 4x ein & aus, Leuchtdioden rot & blau an. Bei Ansteuerung über das Spezial-Menu wir erst Home angefhren, kurze Zeit später die erste Psosition zum Level. Hotend bleibt dann stehe, drei aufsteigende Pieptöne in Reihenfolge. Tastpin fährt nicht aus. Auch Befehle M280 P0 S10 oder M280 P0 S90 über Pronterface bringen den Pin nicht zum Ausfahren. Any ideas?

Datafreak, 2021-06-04:
HI @faitesvosjeux,

du nutzt die Firmware mit "11". Funktioniert bei dir der BLTouch bei der Firmware mit "10"?

Gruß,
Datafreak

faitesvosjeux, 2021-06-04:
Hallo Datafreak,

Vers. "10" auch ausprobiert, damit wird die Nozzle gnadenlos ins Druckbett gedrückt...

Datafreak, 2021-06-04:
Also fährt bei Version "10" der PIN vom BLTouch nicht aus.
Bei Version "11" bekommst du den Fehler.

Zeit dir Pronterface bzw. das Terminal in Octoprint einen Fehler an, wenn die Pieptöne kommen?

faitesvosjeux, 2021-06-04:
Bei Version "10" fährt der Pin auch nicht aus, aber die Nozzle ins Druckbett. Nein, keine Fehler im Pronterface, auch kein Echo, wenn der Befehl zum Absenken manuell abgesetzt wird.

MaGe-1, 2021-06-06:
Hast du den Signalpin vom BLTouch nach Anleitung angeschlossen oder direkt auf das Board geführt`?

faitesvosjeux, 2021-06-07:
Ja, da die Platine meines Hotends etwas unterschieldich ist, entsprechend zwei zusätzliche Pins aufgelötet, die zu D11 (auf Mainborad "S1 beschriftet") führen.

knutwurst, 2021-06-29:
> Ja, da die Platine meines Hotends etwas unterschieldich ist, entsprechend zwei zusätzliche Pins aufgelötet, die zu D11 (auf Mainborad "S1 beschriftet") führen.

Dann liegt da der Fehler. Der pin fährt über den Servo-Port aus. Wenn das Signal oben nicht ankommt, macht der BLTouch nichts. Bei der gemeinsamen Masse ist vermutlich der Spannungsabfall zu groß.

faitesvosjeux, 2021-06-29:
Und wie kann man das dann ggf. heilen?

knutwurst, 2021-06-29:
Indem man wie in der Anleitung ein Kabel legt?

faitesvosjeux, 2021-06-29:
Die Pins sind über ein eigenes Kabel auf das Mainboard an D11 (Meinboard "S1") geführt.

MaGe-1, 2021-06-29:
Ansonsten mal die beta 2 probieren... Fehler ist bei mir ähnlich.

faitesvosjeux, 2021-06-29:
Ich habe beta 1 & Beta 2 probiert, keine Änderung.

knutwurst, 2021-07-30:
Ist das Problem mittlerweile gelöst? In dem [Beginner's Guide](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(Deutsch)) habe ich auch noch mal alle Mainboards aufgelistet zur Identifikation.

closed 2021-09-04 by github-actions[bot]