knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #154: [BUG] (short description)

Stewo93, 2021-06-01:
Hi,
Mein Anycubic i3 Mega mit TMC 2208 Stepper Drivern und einem BLTouch homed die Z-Achse nicht richtig.

1. Ich habe mir die Marlinfirmeware2.0.X herunter geladen und die Entsprechende Projecttask auf meinen Drucker geladen.
2. ich wollte die Achsen ganz normal über den Touchscrenn homen 
3. 
**Expected behavior:  Normaler weise sollten sich beide Z-Spindeln gleichzeitig drehen.

**Actual behavior: Es dreht sich leider nur die Rechte Z-Spindel und die linke bleibt stehen. Sobald die eine Achse auf dem Endschalter steht bleibt der Drucker stehen.
Ich habe den Drucker mit zwei Z-Endschaltern und habe die Version MEGA_TMC_BLT_10 auf selbigen geladen.

Datafreak, 2021-06-04:
HI @Stewo93,

sicher das das du nen alten Anycubic i3 Mega und nicht einen Anycubic i3 Mega **S** hast?
Wenn du einen mit "S" hast, dann musst du MEGA_S_TMC_BLT_10 verwenden.

Gruß,
Datafreak

Stewo93, 2021-06-04:
Hi, 
Ich habe definitiv einen i3 mega. Der mega s hat doch nur einen Titanextruder. Ansonsten sind die beiden Drucker doch gleich oder habe ich da einen Denkfehler!🤔 Das Problem liegt bei mir nicht beim Extruder sondern bei der Kalibrierung der Z- Achse.

knutwurst, 2021-06-29:
@Stewo93 Ich gehe von einem elektrischen Problem aus. 
Wie sind die Motoren auf dem Board verkabelt? Sitzen alle Stecker? Sind die Treiber ok?

Stewo93, 2021-07-01:
Hi ja die Kabel sind alle in Ordnung. Beide Motoren drehen wenn die Z-Achse  sich vor dem Homen anhebt.

knutwurst, 2021-07-30:
Dann vermute ich, dass es sich um ein Problem mit den Endschaltern handelt. Beim homen (und nur da!) werden die Endschalter ausgewertet und danach durch Soft-Endstops ersetzt. 

Wenn das normale fahren klappt, aber das Homen nicht, liegt es an den Endstops.

closed 2021-09-04 by github-actions[bot]